moode-player/moode #496: moode server does not respond over IPv6

kburdett, 2022-02-24:
I have noticed that the moode web UI does not respond to requests via IPv6. I did some digging and noticed that nginx wasn't binding to port 80 for IPv6. The default configuration is binding over IPv4 only. It is an easy change to add a listen directive for the IPv6 port as well. I have submitted #495 to make this change.

I tested this change out locally and it worked great... until the server restarted. After that, I began receiving some checksum errors from the config validation and had to revert the change. I consider that motivation to stop being lazy and submit a PR for the change instead :)

moodeaudio, 2022-02-25:
Whats the usage scenario where IPv6 is useful given that moOde is meant for LAN and not WAN?

Fyi, the checksum stuff was dropped for upcoming moOde 8.

kburdett, 2022-02-25:
I don't suppose there are any usage scenarios not met by IPv4.

I run IPv6 internally, pretty much exclusively. Moode stands out like a sore thumb in my DNS config. Working around the issue isn't difficult, since I am (of course) running dual-stack. Supporting IPv6 is pretty trivial these days, why deploy a service for IPv4 only?

moodeaudio, 2022-02-25:
Makes sense. I'll do some testing and also run it by the crew.

The initial 8 release is more or less frozen but there will be plenty of 8 series releases to come.

moodeaudio, 2022-12-02:
Will reopen if other users request IPv6

closed 2022-12-02 by moodeaudio

marrobHD, 2022-12-19:
I have an eye on this for a while. IPv6 is now days quite common. Would love to see IPv6 support, which shouldn't be too difficult to integrate.

marrobHD, 2022-12-19:
> Will reopen if other users request IPv6

I have an eye on this for a while. IPv6 is now days quite common. Would love to see IPv6 support, which shouldn't be too difficult to integrate.

moodeaudio, 2022-12-19:
Whats the usage scenario where IPv6 is needed?

marrobHD, 2022-12-20:
> 

It's not *needed* in the short term, but it's neat to have. I also use IPv6 internally.