moode-player/moode #498: slowness of thumbnail generator

empierre, 2022-02-27:
Hello,

  I've moved vrom volumio to moode, fantastic ! 

  My only concern so far is that the thumbnails generation takes so long... more than 4 hours yet and no idea when it will finish.

  Any way to check or improve the performance ?

moodeaudio, 2022-02-28:
I only use Git issues for developer related things. Your best bet is to post on the Moode Forum https://moodeaudio.org/forum in the Support section. Email tim at moodeaudio dot org with your preferred userid and I'll create an account for you.

Include the moode startup log in your post using the command below so we know something about your config.
```
moodeutl -l
```

-Tim

closed 2022-02-28 by moodeaudio

empierre, 2022-02-28:
Hello here is the log:

20220227 131442 worker: -- Start
20220227 131442 worker: Successfully daemonized
20220227 131443 worker: Integrity check (passed)
20220227 131445 worker: File check (OK)
20220227 131446 worker: Session vacuumed
20220227 131446 worker: Session loaded
20220227 131446 worker: Debug logging (OFF)
20220227 131447 worker: -- Audio debug
20220227 131447 worker: ALSA cards: (0:IQaudIODAC | 1:empty | 2:empty |
3:empty
20220227 131447 worker: MPD config: (0:iqaudio-dacplus | mixer:(Digital) |
card:0)
20220227 131447 worker: ALSA mixer actual (Digital)
20220227 131447 worker: ALSA Digital volume set to (0%)
20220227 131447 worker: -- System
20220227 131449 worker: Host     (moode)
20220227 131449 worker: moOde    (7.6.1 2021-12-17)
20220227 131449 worker: RaspiOS  (10.11)
20220227 131449 worker: Kernel   (5.10.63-v7+ #1496)
20220227 131449 worker: Platform (Pi-2B 1.1 1GB)
20220227 131449 worker: ARM arch (armv7l, 32-bit kernel)
20220227 131449 worker: MPD ver  (0.23.4_p0x3)
20220227 131449 worker: CPU gov  (ondemand)
20220227 131449 worker: USB boot not available
20220227 131449 worker: File system expanded
20220227 131449 worker: HDMI port on
20220227 131449 worker: -- Network
20220227 131449 worker: IP address check timeout (90 secs)
20220227 131449 worker: eth0 adapter exists
20220227 131449 worker: eth0 IP address check (No)
20220227 131450 worker: eth0 address not assigned
20220227 131450 worker: wlan0 adapter exists
20220227 131450 worker: wifi country (FR)
20220227 131450 worker: wlan0 trying SSID (nef2)
20220227 131450 worker: IP addr (192.168.86.19)
20220227 131450 worker: Netmask (255.255.255.0)
20220227 131450 worker: Gateway (192.168.86.1)
20220227 131450 worker: Pri DNS (192.168.86.1)
20220227 131450 worker: Domain  (lan)
20220227 131450 worker: -- Audio config
20220227 131451 worker: MPD conf updated
20220227 131451 worker: ALSA card number (0)
20220227 131451 worker: MPD audio output (None)
20220227 131451 worker: Audio formats (S16_LE, S24_LE, S32_LE)
20220227 131451 worker: ALSA mixer name (Digital)
20220227 131451 worker: MPD mixer type (software)
20220227 131451 worker: Hdwr volume controller exists
20220227 131451 worker: Max ALSA volume (100%)
20220227 131451 worker: ALSA output mode (Default: plughw)
20220227 131452 worker: ALSA loopback (Off)
20220227 131452 worker: Reset renderer active flags
20220227 131452 worker: CamillaDSP (off)
20220227 131452 worker: -- MPD startup
20220227 131452 worker: MPD started
20220227 131824 worker: MPD accepting connections
20220227 131824 worker: MPD output 1 ALSA Default (on)
20220227 131824 worker: MPD output 2 ALSA Bluetooth (off)
20220227 131824 worker: MPD output 3 HTTP Server (off)
20220227 131824 worker: MPD crossfade (off)
20220227 131824 worker: MPD ignore CUE files (yes)
20220227 131824 worker: -- Feature availability
20220227 131824 worker: Source select (available)
20220227 131824 worker: Source select (source: MPD)
20220227 131824 worker: Source select (output: iqaudio-dacplus)
20220227 131825 worker: Bluetooth pairing agent (started)
20220227 131825 worker: Bluetooth (available: started)
20220227 131825 worker: Airplay renderer (available)
20220227 131825 worker: Spotify renderer (available)
20220227 131825 worker: Squeezelite (available)
20220227 131825 worker: RoonBridge renderer (not installed)
20220227 131825 worker: Multiroom sender (available)
20220227 131825 worker: Multiroom receiver (available)
20220227 131825 worker: UPnP renderer (available: started)
20220227 131825 worker: DLNA server (available: started)
20220227 131825 worker: UPnP browser (available: started)
20220227 131825 worker: GPIO button handler (available)
20220227 131825 worker: Stream recorder (n/a)
20220227 131825 worker: -- Music sources
20220227 131826 worker: USB sources (none attached)
20220227 131833 worker: NAS and UPnP sources (mountall initiated)
20220227 131833 worker: -- Other
20220227 131833 worker: USB volume knob (Off)
20220227 131834 worker: Shellinabox SSH started
20220227 131834 worker: USB auto-mounter (udisks-glue)
20220227 131834 worker: LED0 (Off)
20220227 131834 worker: LED1 (Off)
20220227 131834 worker: Saved MPD vol level (0)
20220227 131834 worker: Preamp volume level (0)
20220227 131834 worker: MPD volume level (7) restored
20220227 131834 worker: ALSA Digital volume (100%)
20220227 131834 worker: Auto-play (Off)
20220227 131834 worker: Maintenance interval (3 hours)
20220227 131834 worker: Screen saver activation (Never)
20220227 131834 worker: Session permissions (OK)
20220227 131834 worker: Watchdog started
20220227 131834 worker: Ready

empierre, 2022-02-28:
Did a quick checks on packages and security (using lynis), some ideas for
hardening a little bit:

In the installed packages there are a lot about:
- X11 + chromium + xfonts
- compilers (gcc...)
- python 2 and 3  (2 is out of support now)
- tcl

are they needed for direct login through HDMI ?

Missing for optimal wifi adapter coverage:
firmware-misc-nonfree
firmware-ralink

Add security updates:
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys
AA8E81B4331F7F50 112695A0E562B32A
/etc/apt/sources.list
deb http://deb.debian.org/debian-security/ buster/updates main

chmod 027 /etc/login.defs

/etc/ssh/sshd_config:
AllowTcpForwarding no
ClientAliveCountMax 2
Compression no
MaxAuthTries 2
MaxSessions 2
PermitRootLogin no
TCPKeepAlive no
X11Forwarding no
AllowAgentForwarding no

empierre, 2022-02-28:
Hello,

  more than 12 hours as updating thumbnails, restarted the pi and have this
after forcing afterthumbnails refresh:

0220228 102616 worker: Saved MPD vol level (0)
20220228 102616 worker: Preamp volume level (0)
20220228 102616 worker: MPD volume level (7) restored
20220228 102617 worker: ALSA Digital volume (100%)
20220228 102617 worker: Auto-play (Off)
20220228 102617 worker: Maintenance interval (3 hours)
20220228 102617 worker: Screen saver activation (Never)
20220228 102617 worker: Session permissions (OK)
20220228 102617 worker: Watchdog started
20220228 102617 worker: Ready
20220228 102747 worker: Job regen_thmcache
20220228 102747 thmcache: Start
20220228 102747 thmcache: Priority: Embedded cover
20220228 102747 thmcache: Res,Qual: Auto
20220228 102747 thmcache: Px ratio: 1
20220228 102747 thmcache: Th width: 100
20220228 102747 thmcache: Thm qual: 75
20220228 102747 thmcache: Scanning: NAS, SDCARD, UPNP

empierre, 2022-02-28:
Hi,

  Seems scanning is effective now.

20220228 110111 thmcache: mp3: Zend media exception: File does not contain ID3v2 tag
20220228 110112 thmcache: mp3: /var/lib/mpd/music/NAS/musicsmb/Stream/a/stream.rm.mp3
20220228 110112 thmcache: mp3: Zend media exception: File does not contain ID3v2 tag
20220228 110113 thmcache: mp3: /var/lib/mpd/music/NAS/musicsmb/Stream/c/stream.rm.mp3
20220228 110113 thmcache: mp3: Zend media exception: File does not contain ID3v2 tag
20220228 111028 thmcache: Done: 7705 folders scanned, 6491 thumbs created, 1214 already in cache.

  It seems that first connection to mount after reboot generates mpd error, and maybe a status issue has blocked indexing images.