moode-player/moode #499: Add built-in support for AAC, AptX, AptX HD and LDAC.

the4anoni, 2022-03-07:
Hi
https://moodeaudio.org/forum/showthread.php?tid=3683&highlight=aptx Can we have this built-in?

In terms of licence: FDK-AAC is open source and It is free, OpenAptX is licenced under MIT Licence.
I wanted to say this on forum, but I couldn't make issue there.

moodeaudio, 2022-03-07:
From the README in openaptx repo https://github.com/Arkq/openaptx
```
"This project is for research purposes only. Without a proper license private 
and commercial usage might be a case of a patent infringement."
```

From the fdx-aac Wiki https://en.wikipedia.org/wiki/Fraunhofer_FDK_AAC
```
"The license included by Fraunhofer in the FDK library source code
[[17]](https://en.wikipedia.org/wiki/Fraunhofer_FDK_AAC#cite_note-:1-17) allows 
redistribution in source or binary forms, but does not license patented technologies
described by the code. The license states that the library may only be used for
purposes as authorized by patent licenses. Due to this restriction, along with
 a limitation on charging for the library, [Debian](https://en.wikipedia.org/wiki/Debian) 
considers it non-free."
```
From ffmpeg https://trac.ffmpeg.org/wiki/Encode/AAC
```
"The license of libfdk_aac is not compatible with GPL, so the GPL does not permit
distribution of binaries containing incompatible code when GPL-licensed code is 
also included. Therefore this encoder have been designated as "non-free", and 
you cannot download a pre-built ffmpeg that supports it."
```

closed 2022-03-07 by moodeaudio

moodeaudio, 2022-03-07:
The codecs have to be compiled into the bluez-alsa backend component https://github.com/Arkq/bluez-alsa which due to the restrictive licenses associated with the codecs would prevent moOde from including a Bluetooth audio feature.

You best bet is to create your own custom compiled bluez-alsa component along with an installer that people can download from your repo.

the4anoni, 2022-03-08:
> The codecs have to be compiled into the bluez-alsa backend component https://github.com/Arkq/bluez-alsa which due to the restrictive licenses associated with the codecs would prevent moOde from [...]

I compiled bluez-alsa using this manual https://moodeaudio.org/forum/showthread.php?tid=3683&highlight=aptx and bluealsa -h outputs:
`Available BT audio codecs:
  a2dp-source:  SBC, AAC, aptX, aptX-HD, LDAC
  a2dp-sink:    SBC, AAC, aptX, aptX-HD
  hfp-*:        CVSD`

but both bluealsa-aplay -L and my android phone (Galaxy S10) shows that I am using AAC, and I can't select AptX under developer settings. What I am doing wrong?