moode-player/moode #500: ROTEL A12 pc-usb do not work

eskimos2000, 2022-03-09:
Hi there,

I want to enjoy my music via RP4 connected with USB to PC-USB in Rotel A12. I manage to make it works by setting also to card 2 according to play:
card 2: R20 [Rotel USB Audio 2.0], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

I set MPD to Rotel USB Audio 2.0 as it was an option. I set Rotel PC-USB to USB 1.0 (so only 96khz) as 2.0 (192khz) cause MPD error can't open the device.
Now seems to ono errors but still no sound at all.

`Audio Information
INPUT / OUTPUT
Sourcehttp://ice1.somafm.com/beatblender-128-aac
Encoded atVBR compression
Decoded to16 bit, 44.1 kHz, Stereo, 128 kbps
Output rate24 bit, 44.1 kHz, Stereo, 2.117 Mbps
Output modeALSA Default (plughw)
Output chainMPD -> plughw -> DeviceDSP OPERATIONS
Resample rateOff
Selective rateOff
Resample qualityOff
CamillaDSPOff
Volume typeSoftware (MPD)AUDIO DEVICE
DeviceRotel USB Audio 2.0
Mixer nameRotel USB Audio 2.0
InterfaceUSB
FormatsS24_3LE
PlatformPi-4B 1.1 2GB`

Any suggestion, please?

eskimos2000, 2022-03-09:
moode log

Moode Log
			
20220309 203201 worker: -- Start
20220309 203201 worker: Successfully daemonized
20220309 203201 worker: Integrity check (passed)
20220309 203201 worker: File check (OK)
20220309 203202 worker: Session vacuumed
20220309 203202 worker: Session loaded
20220309 203202 worker: Debug logging (OFF)
20220309 203202 worker: -- Audio debug
20220309 203202 worker: ALSA cards: (0:b1 | 1:Headphones | 2:R20 | 3:empty
20220309 203202 worker: MPD config: (2:Rotel USB Audio 2.0 | mixer:(Rotel USB Audio 2.0 ) | card:2)
20220309 203202 worker: ALSA mixer actual (Rotel USB Audio 1.0 )
20220309 203202 worker: ALSA Rotel USB Audio 1.0 volume set to (0%)
20220309 203202 worker: -- System
20220309 203203 worker: Host     (moodeserver)
20220309 203203 worker: moOde    (7.6.1 2021-12-17)
20220309 203203 worker: RaspiOS  (10.6)
20220309 203203 worker: Kernel   (5.4.77-v8+ #1371)
20220309 203203 worker: Platform (Pi-4B 1.1 2GB)
20220309 203203 worker: ARM arch (aarch64, 64-bit kernel)
20220309 203203 worker: MPD ver  (0.23.4_p0x3)
20220309 203203 worker: CPU gov  (ondemand)
20220309 203203 worker: USB boot not available
20220309 203203 worker: File system not expanded yet
20220309 203203 worker: HDMI port off
20220309 203203 worker: -- Network
20220309 203203 worker: IP address check timeout (90 secs)
20220309 203203 worker: eth0 adapter exists
20220309 203203 worker: eth0 IP address check (No)
20220309 203203 worker: IP addr (192.168.1.200)
20220309 203203 worker: Netmask (255.255.255.0)
20220309 203203 worker: Gateway (192.168.1.1)
20220309 203203 worker: Pri DNS (192.168.1.1)
20220309 203203 worker: Domain  (lan)
20220309 203203 worker: wlan0 adapter does not exist
20220309 203203 worker: -- Audio config
20220309 203203 worker: MPD conf update skipped (USB device)
20220309 203203 worker: ALSA card number (2)
20220309 203203 worker: MPD audio output (Rotel USB Audio 2.0)
20220309 203204 worker: Audio formats (S24_3LE)
20220309 203204 worker: ALSA mixer name (Rotel USB Audio 2.0 )
20220309 203204 worker: MPD mixer type (software)
20220309 203204 worker: Hdwr volume controller not detected
20220309 203204 worker: ALSA output mode (Default: plughw)
20220309 203204 worker: ALSA loopback (Off)
20220309 203204 worker: Reset renderer active flags
20220309 203204 worker: CamillaDSP (off)
20220309 203204 worker: -- MPD startup
20220309 203204 worker: MPD started
20220309 203204 worker: MPD accepting connections
20220309 203204 worker: MPD output 1 ALSA Default (on)
20220309 203204 worker: MPD output 2 ALSA Bluetooth (off)
20220309 203204 worker: MPD output 3 HTTP Server (off)
20220309 203204 worker: MPD crossfade (off)
20220309 203204 worker: MPD ignore CUE files (yes)
20220309 203204 worker: -- Feature availability
20220309 203204 worker: Source select (available)
20220309 203204 worker: Source select (source: MPD)
20220309 203204 worker: Source select (output: Rotel USB Audio 2.0)
20220309 203204 worker: Bluetooth (available)
20220309 203204 worker: Airplay renderer (available)
20220309 203204 worker: Spotify renderer (available)
20220309 203204 worker: Squeezelite (available)
20220309 203204 worker: RoonBridge renderer (not installed)
20220309 203204 worker: Multiroom sender (available)
20220309 203204 worker: Multiroom receiver (available)
20220309 203204 worker: UPnP renderer (available)
20220309 203204 worker: DLNA server (available)
20220309 203204 worker: UPnP browser (available)
20220309 203204 worker: GPIO button handler (available)
20220309 203204 worker: Stream recorder (n/a)
20220309 203204 worker: -- Music sources
20220309 203204 worker: USB source (7b952e33-7045-4155-9865-e7962f56faa4)
20220309 203204 worker: NAS and UPnP sources (none configured)
20220309 203204 worker: -- Other
20220309 203204 worker: USB volume knob (Off)
20220309 203204 worker: USB auto-mounter (udisks-glue)
20220309 203204 worker: LED0 (On)
20220309 203204 worker: LED1 (On)
20220309 203204 worker: Saved MPD vol level (0)
20220309 203204 worker: Preamp volume level (0)
20220309 203204 worker: MPD volume level (1) restored
20220309 203204 worker: ALSA volume level (None)
20220309 203204 worker: Auto-play (Off)
20220309 203204 worker: Maintenance interval (3 hours)
20220309 203204 worker: Screen saver activation (Never)
20220309 203204 worker: Session permissions (OK)
20220309 203205 worker: Watchdog started
20220309 203205 worker: Ready

moodeaudio, 2022-03-09:
I only use Git issues for development related things but you can get support over at the Moode Forum. Email tim at moodeaudio dot org with your preferred userid and I'll setup an account for you.

closed 2022-03-09 by moodeaudio