moode-player/moode #506: Moode player in docker container

MaxPayne86, 2022-03-30:
Hello, I've just seen some very shy requests on the forum. I'm running a very brand new Arduino Portenta-X8 which is a base yocto distro + docker-compose apps. BalenaOS users (Rasperry Pi) may benefit too since the concept is the same.

In short, I don't want an entire new distro image for just stream audio to a speaker. I already have a secure and optimized yocto distro for my embedded system which among other tasks handle ota updates and cloud communication. I'm looking for a way to implement moode player functionalities in a docker-compose app.

```
docker run -it --privileged debian:bullseye-slim /bin/bash
$ apt-get update
$ apt-get install -y curl gnupg
$ curl -fsSL 'https://dl.cloudsmith.io/public/moodeaudio/m8y/gpg.E5A251A46C58117E.key' | gpg --dearmor | tee /etc/apt/trusted.gpg.d/E5A251A46C58117E.gpg > /dev/null
$ curl -fsSL 'https://dl.cloudsmith.io/public/moodeaudio/m8y/config.deb.txt?distro=raspbian&codename=bullseye' > /etc/apt/sources.list.d/moodeaudio-m8y.list
$ apt-get update
Hit:1 http://deb.debian.org/debian bullseye InRelease
Hit:2 http://deb.debian.org/debian bullseye-updates InRelease                                
Hit:3 http://security.debian.org/debian-security bullseye-security InRelease                 
Get:4 https://dl.cloudsmith.io/public/moodeaudio/m8y/deb/raspbian bullseye InRelease [5167 B]
Get:5 https://dl.cloudsmith.io/public/moodeaudio/m8y/deb/raspbian bullseye/main Sources [10.4 kB]
Get:6 https://dl.cloudsmith.io/public/moodeaudio/m8y/deb/raspbian bullseye/main arm64 Packages [4449 B]
Fetched 20.0 kB in 2s (9226 B/s)   
Reading package lists... Done

So basically this app is packetized for raspbian not "debian", so the following is somewhat expected

$ apt-get install moode-player=8.0.2-1moode1
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 moode-player : Depends: rpi-update but it is not installable
                Depends: djmount but it is not installable
                Depends: bluez-firmware but it is not installable
                Depends: pi-bluetooth but it is not installable
                Depends: ashuffle but it is not installable
                Depends: alsa-cdsp but it is not installable
                Depends: alsacap but it is not installable
                Depends: bluez-alsa-utils but it is not installable
                Depends: libasound2-plugin-bluez but it is not installable
                Depends: camilladsp but it is not installable
                Depends: librespot but it is not installable
                Depends: python3-libupnpp but it is not installable
                Depends: udisks-glue but it is not installable
                Depends: upmpdcli but it is not installable
                Depends: chromium-browser but it is not installable
E: Unable to correct problems, you have held broken packages.
```
I'm abandoning the deb package in favour of a build from sources. Do you have any clue on the minimal dependencies I need for basic moode functionality like stream local fs audio to an i2s codec? Then we could adjust bluetooth and the remaining stuff. What about chromium-browser dep? Is it strictly required?

Thanks

moodeaudio, 2022-03-30:
I'm not familiar with the platform you are using and so am not able to help you out but as far as dependencies goes, all of them are required in our build because all the features and options they support are used by various segments of our user community.

closed 2022-03-30 by moodeaudio

chourmovs, 2023-08-20:
> Hello, I've just seen some very shy requests on the forum. I'm running a very brand new Arduino Portenta-X8 which is a base yocto distro + docker-compose apps. BalenaOS users (Rasperry Pi) may [...]

> I'm abandoning the deb package in favour of a build from sources. Do you have any clue on the minimal dependencies I need for basic moode functionality like stream local fs audio to an i2s codec? [...]

I spent a lot of time building something similar for my ASUS tinker board
https://github.com/chourmovs/moode_tinkerboard
Look at the recipe .sh, maybe you could test/adapt

Good luck

moodeaudio, 2023-08-20:
Sure, I'll check it out but I'm not too familiar with Docker. As far as our project goes we focus on the Raspberry Pi family of SBC's and the associated software and support ecosystem. The challenge with spending time on Raspberry Pi alternatives is that they don't seem to really offer a massive improvement and they lack the huge Pi software/support ecosystem.

chourmovs, 2023-08-20:
Hey Tim, thank you for your answer.

I just posted an answer to the OP to let a trace in Google cache for future person interested. I don't wait from you that you spend time on this topic, this is quite exotic exploit using docker to answer to very specific situation (Mine is that I don't want to emmc flash moode on my volumio primo to not loose my lifetime license and warranty)

you do a fantastic job with moode 👍, Keep it up

chourmovs, 2023-08-20:
and I forget to mention that my script is working, everything is working except camilladsp in fact 🎉

moodeaudio, 2023-08-20:
Nice. I think CamillaDSP uses some http ports so maybe they are being blocked by the Docker environment.