moode-player/moode #507: Improvements for radio "DR P8 Jazz" [id=35]

lbschenkel, 2022-04-01:
- Add homepage: https://www.dr.dk/lyd/p8jazz
- Current stream format is (192 kbps) MP3, not AAC
- There is an even better stream (320 kbps AAC): https://DRliveradio.akamaized.net/hls/live/2022411/p8jazz/playlist-320000.m3u8

moodeaudio, 2022-04-02:
Thx, I'll include the updates in upcoming 8.1.0

closed 2022-04-02 by moodeaudio

lbschenkel, 2022-04-02:
One more thing that I forgot: note that the 320 kbps AAC stream does not include metadata, unlike the 192 kbps MP3 one. Therefore you might want to include the entry as a different radio (like `DR P8 Jazz (320K)`), like it was done for BBC.

moodeaudio, 2022-04-02:
The home page URL was meant to be used when clicking on the Title named "Streaming source" which is displayed when a station does not transmit an a Title tag.

The DR stations either transmit an actual Title tag or they transmit the station URL thus clicking on the Title in the UI will always result in a Google search.

lbschenkel, 2022-04-02:
Sure, just mentioning that when using the MP3 stream the user will see the current song in the UI, but not when using the AAC one. This may be a regression that you consider important. Personally I think that the increase in audio quality is worth the tradeoff, but I am not the maintaner of this project so it is not my call.

moodeaudio, 2022-04-02:
No prob. I can add the 320K stream for upcoming 8.1.0