moode-player/moode #508: No sound through IQadIO Pi-DigiAMP+ since update to version 8.0.2

aq18507, 2022-04-03:
Since updating to version 8.0.2, I am unable to get any sound through the IQadIO Pi-DigiAMP+ attached to a Raspberry Pi Zero w. The system has been working absolutely fine for about two years. I am running 8 Players across the entire house, and all of them are updated without any issues apart from this particular one.

What I have done so far
-  Of course I checked whether the module itself is broken by running it of a Pi 4 8GB and Pi 3 on version 8. It works fine with the two larger Pi's.
- I have also tried it with a brand new Pi Zero 2 (Unused new out of the package), and I haven't managed to get it going either.
- I have reverted the Pi in question to version 7.6.1 to realise that this doesn't work anymore.
- Ther is one other instance running, and updated with the same setup, and this is working, which I find weird.
- It appears to have a connection to the IQadIO Pi-DigiAMP+, which I have checked over the command line.
- The mute LED is off on the IQadIO Pi-DigiAMP+ board.
- I even tried loading Volumio onto it and it didn't work with it either.

I am wondering where this is a Kernel related issue here?

moodeaudio, 2022-04-03:
Your best bet is to post in the support section of our Forum. If you need a Forum account email tim at moodeaudio dot org with your preferred userid and I'll create the account.

closed 2022-04-03 by moodeaudio

aq18507, 2022-04-03:
Thank you very much and sorry for spamming this page - this wasn't my intention. I indeed tried to post there but couldn't. I shall send you an email straight away.