moode-player/moode #518: Moode audio Control im home assistant.

kuduacz, 2022-05-22:
I have a problem with control moode audio from HA.
I set metadata file to on.
Mirror link it to /var/www/ by   sudo ln /var/local/www/currentsong.txt /var/www/currentsong.txt command.

And IT work till i change render. I mean when i play domek internet radio and change statoion mirrored file have no change. 
I try to use that file to better Control IT by Home assistant to see actual render like BT or airplay.
Add sensor on HA to see actual render.

moodeaudio, 2022-05-22:
Your best bet is to post the issue in the support section of our Forum. Email tim at moodeudio dot org with your preferred userid and I'll setup an account for you.

closed 2022-05-22 by moodeaudio

rvandet, 2023-04-14:
Did you progress on your issue?

I would like to control all sort of music sources (Spotify, MPD, iTunes, SMBv1) from HA.
iTunes offcourse is a no go yet, but the others should be aware of what is playing making it wife-friendly.

kuduacz, 2023-06-27:
unfortunetaly not, still no progress.