moode-player/moode #522: Build libmpdclient from sources

TheOldPresbyope, 2022-06-04:
From time to time, the MPD protocol and MPD-database API have new metadata items added. This affects code in both MPD and the helper library libmpdclient. 

It appears that moOde installs the libmpdclient-dev package from the Raspberry Pi OS repo, which as of today stands at v2.19-1. I propose we instead build libmpdclient from sources. The latest source-code release is v2.20 while the unreleased HEAD currently stands at v2.21. Recently, each bump in version has brought along some new metadata tags defined in the MPD docs.

This change would not affect the behavior of moOde today but it would allow intrepid users to exploit the new tags defined in the MPD docs. A case in point is the newest tag "MOOD" which was mentioned in a recent forum thread.

bitkeeper, 2022-06-25:
Maybe a good point to do this, is the next tim we also upgrade MPD.
The build of MPD pacakge also depends on libmpdclient2.

Luckily Debian already provides the 2.20 version, at least when testing repo of bookworm is used.
So it can be directly used or rebuild from source package.

Refs:
* https://packages.debian.org/bookworm/libmpdclient2 
* https://packages.debian.org/bookworm/libmpdclient-dev

moodeaudio, 2023-11-14:
It sounds like we just go with whats in the OS repo.

Will reopen if needed.

closed 2023-11-14 by moodeaudio