moode-player/moode #525: Error in librespot SQL table name in release 8.1.0

kionez, 2022-06-24:
Hi, after updating to 8.1.0 I noticed a issue with Spotify. When is enabled the UI goes unresponsive ( same issue in [this forum post](https://moodeaudio.org/forum/showthread.php?tid=4900) ).

After digging into the code i discover that in _inc/renderer.php_ at line 87 there is:
`$result = sqlRead('cfgSpotify', sqlConnect());`
but in the sqlite database the table name is cfg_spotify, so i modified in:
`$result = sqlRead('cfg_spotify', sqlConnect());`
and everything works fine.

Also, there is a typo in _util/autocfg-import.php_ at line 32, there is a `)` missing.

Hope it helps.

```
$ moodeutl -s
Gathering info...

S Y S T E M   P A R A M E T E R S

moOde release           = 8.1.0 2022-MM-DD
RaspiOS                 = 11.2
Linux kernel            = 5.15.32-v7+ #1537
Platform                = Pi-3B+ 1.3 1GB
Architecture            = armv7l (32-bit)
System uptime           = up 1 hour, 40 minutes
Timezone                = Europe/Rome
Current time            = 2022-06-24 09:03:56
```

kionez, 2022-06-24:
I forgot to mention that because of this error the worker.php crashes.

moodeaudio, 2022-06-24:
Many thanks! I'll fix :-)

closed 2022-06-27 by kionez