moode-player/moode #527: Image for AllWinner A10

arianaglande, 2022-07-03:
Hey, 
is it possible for you to produce a bootable image for the AllWinner A10 SoC (https://linux-sunxi.org/A10)? It's armv7 32 bit, i have 29 tablets with that cpu and wanted to do smth cool with it, if i put a microsd card with a valid bootloader it boots straight away but it doesnt with your image, it boots android instead

 thx

moodeaudio, 2022-07-03:
Our project only supports Raspberry Pi OS and the Raspberry Pi family of SBC's.

closed 2022-07-03 by moodeaudio

arianaglande, 2022-07-03:
> Our project only supports Raspberry Pi OS and the Raspberry Pi family of SBC's.

but if i take another bullseye image and put all the files in your repo it should work right? considering a generic armv7 cpu