moode-player/moode #532: can't turn off integrated WiFI adapter on Pi-Zero 2W

TheOldPresbyope, 2022-07-21:
There is no option offered on the sys config screen to turn off the integrated WiFi adapter on my Pi-Zero 2W 

Lines 340-341 in www/sys-config.php
```
// Pi-Zero W, Pi-3B/B+/A+, Pi-4B
if ($name == 'Pi-Zero W 1.1 512MB' || $model == '3' || $model == '4') {
```

need to be amended to include the Pi-Zero 2W (so does the information popup)

The output from pirev.py for mine is
```
pi@m81232:~ $ /var/www/util/pirev.py
0x902120	Zero 2 W	1.0	512MB	Sony UK	BCM2837
```

moodeaudio, 2022-07-21:
Fixed :-)

TheOldPresbyope, 2022-07-22:
Ooh, so close ;)

```
if (stripos($name, 'Pi-Zero W' !== false) || stripos($name, 'Pi-Zero 2 W' !== false) || $model == '3' || $model == '4') {
```

The parentheses are misplaced in the stripos() calls.

After repairing this, testing was successful in both a Zero W and a Zero 2 W.

moodeaudio, 2022-07-22:
lol, never rely on the fact the the code compiles cleanly :-0

moodeaudio, 2022-07-22:
Fixed :-)