moode-player/moode #533: 500 on net-config.php page if SSID has `'` in it

morgothulhu, 2022-07-23:
Repro:
- wireless LAN setup, SCAN, pick SSID with `'` in it
- save
- navigate to network preferences page again

Expected:
- can see page

Actual:
- 500 error

Looking at php errors - seems related to bad `sqlQuery`. Maybe https://github.com/moode-player/moode/blob/develop/www/net-config.php#L136 not escaping `'` in SSID?

moodeaudio, 2022-07-23:
Thanks, I'll investigate.

moodeaudio, 2022-07-23:
I've updated net-config.php. There were several statements where ssid was not being escaped. Give it a try and let me know if it resolves your issue.
https://github.com/moode-player/moode/commit/1de94d0801f7dd5e8781bdf96bb74d39f1d0b021

morgothulhu, 2022-07-23:
@moodeaudio I had to also escape L136 in addition to get the page to render - but now (after reboot) the SSID displayed is double-escaped (e.g. my SSID is `ab'c` but I see `ab''c`). 

The wifi network cannot connect.

moodeaudio, 2022-07-24:
Right, it was being double escaped. The second escape was in inc/sql.php function sqlUpdate().

This latest commit https://github.com/moode-player/moode/commit/641c1396533000d4c92b83eab25af0021e2b98ec should do the trick. You might have to clean up tables cfg_ssid and or cfg_network before testing.

moodeaudio, 2022-07-26:
Will reopen if still broken.