moode-player/moode #534: Purposal for a possible djmount utility replacement

asdavidov, 2022-07-30:
In a recent versions of Moode audio player the djmount utility was removed and the DLNA/UPnP server browsing functionality is now unavailable.
But due to the fact that DLNA protocol is actually a SOAP over HTTP, it is possible to replace the djmount utility with the standard Unix/Linux utilities and a simple bash or PHP scripting.
An example of such scripting effort could be found at the https://github.com/javier-lopez/learn/blob/master/sh/tools/simple-dlna-browser project with a usage description posted at the author's homepage at http://javier.io/blog/en/2016/01/22/simple-upnp-dlna-browser.html
Such scripting is not a one-to-one djmount (or DLNA server mounting via FUSE) replacement, but it'll allow to search for and query DLNA servers to get information from them that could be used further within the mpd player.

So this proposal is the follow-up to this https://moodeaudio.org/forum/showthread.php?tid=4974 forum thread.

moodeaudio, 2022-07-31:
I think it would be quite a challenge to integrate such a script but MPD does include a UPnP database plugin https://mpd.readthedocs.io/en/latest/plugins.html#upnp which I think allows using MPD commands such as `lsinfo` to browse UPnP media servers. ??

moOde's Folder view already uses these commands so maybe the `upnp` database plugin with some  amount of configuration would result in UPnP media servers showing up in Folder view.

The MPD package build used in moOde 8 series doesn't include this plugin but you could try manually compiling MPD with the `upnp` plugin by using a section from our old build recipe. I haven't tested this but it looks like it should work.

```
#
# Install dev libs
#

sudo apt -y install \
libyajl-dev \
libasound2-dev \
libavahi-client-dev \
libavcodec-dev \
libavformat-dev \
libbz2-dev \
libcdio-paranoia-dev \
libcurl4-gnutls-dev \
libfaad-dev \
libflac-dev \
libglib2.0-dev \
libicu-dev \
libid3tag0-dev \
libiso9660-dev \
libmad0-dev \
libmpdclient-dev \
libmpg123-dev \
libmp3lame-dev \
libshout3-dev \
libsoxr-dev \
libsystemd-dev \
libvorbis-dev \
libwavpack-dev \
libwrap0-dev \
libzzip-dev \
libpcre++-dev

#
# Install build tooling
#

sudo apt -y install meson

#
# Download MPD sources
#

BRANCH=0.23
VERSION=.5
sudo wget http://www.musicpd.org/download/mpd/$BRANCH/mpd-$BRANCH$VERSION.tar.xz
sudo tar xf mpd-$BRANCH$VERSION.tar.xz
cd mpd-$BRANCH$VERSION
sudo meson . output/release --buildtype=release -Db_ndebug=true

#
# Configure compile options.
#

sudo meson configure \
-Dalsa=enabled \
-Dbzip2=enabled \
-Dcurl=enabled \
-Ddatabase=true \
-Ddsd=true \
-Dffmpeg=enabled \
-Dfaad=enabled \
-Dflac=enabled \
-Dhttpd=true \
-Did3tag=enabled \
-Dlame=enabled \
-Dlibmpdclient=enabled \
-Dmad=enabled \
-Dmpg123=enabled \
-Dpipe=true \
-Drecorder=true \
-Dshout=enabled \
-Dsoxr=enabled \
-Dvorbis=enabled \
-Dwave_encoder=true \
-Dwavpack=enabled \
-Dzeroconf=avahi \
-Dpcre=enabled \
-Dzzip=enabled \
-Dao=disabled \
-Daudiofile=disabled \
-Ddbus=disabled \
-Dexpat=disabled \
-Dfluidsynth=disabled \
-Dgme=disabled \
-Dipv6=disabled \
-Djack=disabled \
-Dlibsamplerate=disabled \
-Dnfs=disabled \
-Doss=disabled \
-Dpulse=disabled \
-Dsidplay=disabled \
-Dsmbclient=disabled \
-Dsndfile=disabled \
-Dsqlite=disabled \
-Dudisks=disabled \
-Dupnp=disabled \
-Dwildmidi=disabled \
-Dsoundcloud=disabled \
-Dtidal=disabled \
-Dqobuz=disabled
output/release

#
# Compile MPD
#

sudo ninja -C output/release

#
# End
#
```

asdavidov, 2022-07-31:
> I think it would be quite a challenge to integrate such a script but MPD does include a UPnP database plugin https://mpd.readthedocs.io/en/latest/plugins.html#upnp which I think allows using MPD commands such as `lsinfo` to browse UPnP media servers. ??

I thought that djmount was used in moOde audio player with the "-o playlists" option, and now I see that it was a wrong assumption. So that scripting approach was based on it.
I totally agree that if djmount was previously used without that option it would be a big challenge to incorporate such scripting approach in moOde.

Looking at the mpd-upnp historical page at https://www.lesbonscomptes.com/pages/mpd-upnp.html it becoming clear to me that you are right and building MPD with UPnP database plugin support will allow it to search for and play files from the UPnP/DLNA servers almost "automagically".
 
> moOde's Folder view already uses these commands so maybe the `upnp` database plugin with some amount of configuration would result in UPnP media servers showing up in Folder view.

Looking at the https://www.lesbonscomptes.com/pages/mpd-upnp.html#_configuration_file it seems that there is not a lot of options to configure at all ;-)
And in "Trying it out" section there is a description of how the UPnP/DLNA servers and their content should appear - "Once mpd is running, your client should see a list of the network Media Servers as the top level directory."
So it really looks like that building MPD with upnp support should be sufficient to allow it to discover and access UPnP/DLNA servers and their file contents.
Though there are some limitations exists that are also mentioned in that document.
 
> The MPD package build used in moOde 8 series doesn't include this plugin but you could try manually compiling MPD with the `upnp` plugin by using a section from our old build recipe. I haven't tested this but it looks like it should work.

It looks like that libupnp and libexpat development packages are also required to build MPD with UPnP support.
And it seems that curl support should also be enabled (--enable-curl) for mpd-upnp module to work properly.

I'll be on vacation next week so I'll be unable to test build MPD with upnp support quickly, but I'll try to build it when I come back using the instructions that you've provided.

moodeaudio, 2022-07-31:
That would be great :-)

moodeaudio, 2022-12-02:
Will reopen if OP still has interest.

closed 2022-12-02 by moodeaudio