moode-player/moode #535: cmd=get_currentsong doesn't work, and cmd=currentsong doesn't either

dheijl, 2022-07-31:
Version: Moode 8.1.2 2022-07-08

- `get_currentsong` doesn't work, because /var/local/www/currentsong.txt is always empty, I have not tried to figure out why, but I made sure that it never changed once when playing internet radio for hours
- directly sending the `currentsong `cmd to mpd doesn't work either, because of a bug in index.php:
```php
                // MPD commands
                if (false === ($sock = openMpdSock('localhost', 6600))) {
                        workerLog('command/index.php: Connection to MPD failed');
                } else {
                        sendMpdCmd($sock, $_GET['cmd']);
                        $resp = readMpdResp($sock);
                        closeMpdSock($sock);
                        if (stripos($resp, 'Error:')) {
                                echo $resp;
                        }
                }
```
Adding an `else` made `currentsong` work again (although it's perhaps not the best solution):
```php
                // MPD commands
                if (false === ($sock = openMpdSock('localhost', 6600))) {
                        workerLog('command/index.php: Connection to MPD failed');
                } else {
                        sendMpdCmd($sock, $_GET['cmd']);
                        $resp = readMpdResp($sock);
                        closeMpdSock($sock);
                        if (stripos($resp, 'Error:')) {
                                echo $resp;
                        } else {
                                echo $resp;
                        }
```

I should add that switching from Volumio (free) to Moode some months ago has proven to be a smart decision!

moodeaudio, 2022-07-31:
I'm not sure what you mean by `get_currentsong` command doesn't work. If the file is empty it just returns an empty json array to the Browser/caller i.e., `[]`

The MPD section of command/index.php is not designed to return command results to the caller unless it's an error. It's used by the front-end `sendMpdCmd(cmd, async)` function and only for MPD commands that control playback, toggle states and a few others. Commands like `currentsong` are done in the back-end. See `inc/mpd.php`.

What usage scenario are you trying to achieve?

dheijl, 2022-07-31:
I'm trying to get the currently playing song (internet radio in this case) in mpd from the web API.

As ?cmd=get_currentsong always returns an empty json array, I thought I'd use the mpd command directly, as documented in the setup.txt under 2:

COMMAND API
Below are a list of commands that can be submitted to moOde via http or ssh.
1. Base URL is http://moode/command/?cmd=
2. MPD commands that are listed in MPC help

So setup.txt is (no longer) correct and I'd better write my own script using mpc or a socket connection to mpd directly?

moodeaudio, 2022-07-31:
Did you turn on "Metadata file" in System Config?

Heres the help text for the option:
```
Maintain enhanced metadata for the currently playing song or station in the /var/local/www/currentsong.txt file.
NOTE: This file is updated within three seconds after any metadata updates on the UI.
```

dheijl, 2022-07-31:
No I didn't, thanks for pointing this out!
I can see why this is not set by default. 
I searched google on currentsong.txt but got no relevant information.

closed 2022-07-31 by dheijl