moode-player/moode #538: tif and gif cover files not handled in moOde 8.1.2

TheOldPresbyope, 2022-08-09:
After a recent exchange in a support thread on the Forum I put together a test case of coverart files. Since coverart.php suggests image types GIF, JPG, PNG, TIF are all acceptable I created 8 directories

- cover-gif: contains an image file named cover.gif and a test flac file with metadata tag title=cover-gif and no embedded albumart
- cover-jpg: similarly but with "jpg" instead of "gif" everywhere
- cover-png: and so on
- cover-tif: and so forth

- fish-gif: rinse and repeat with an image file named fish.gif
- fish-jpg:...
- fish-png:...
- fish-tif:...

When I update moOde 8.1.2 with these directories under /mnt/SDCARD/CoverFileTests/ I get expected behavior only with the "albums" involving JPG and PNG coverart. The respective image file is displayed in the Library view and the playback view.

For either directory involving GIF coverart, I see, instead, the default moOde image.

For either directory involving TIF coverart, I see, instead, a small black square in the Library view and the "Coverart not found" message in the Playback view.

I don't see anything useful in the logs even with debug logging enabled. I'll have to stick some probes in coverart.php and try again.

Gulping today's develop branch of the moOde github repo, I am unable to get coverart to display at all in moOde 8.2.0 2022-MM-DD. I may be getting ahead of myself.

Sidebar: why the funky coverart file names? I started with my favorite goldfish image and generated copies in various formats, each with the name of the file written into the image. Imagemagick is an amazing suite of tools!

moodeaudio, 2022-08-09:
Odd about tiff format issues. I have a couple of tiff covers in one of my test collections so I'll see if I can repro. 

Have a look at thumb-get.php. It's tedious to debug but there are //workerLog() statements in many code sections that can show what might be going on.

moodeaudio, 2022-08-09:
Quick debug and it looks like width and height are 0 after trying to convert the image to internal format for resizing. The PHP function that does this doesn't support TIFF. https://www.php.net/manual/en/function.imagecreatefromstring.php

I suppose I should add an error message to thumb-gen.php when a format conversion results in 0 width/height. Also should prolly take tif/tiff out of the list of supported cover file formats. 

I think tiff might work if it's embedded though.

moodeaudio, 2022-08-09:
Try latest commit to thumb-get.php. I added an err msg for 0 w/h case.

TheOldPresbyope, 2022-08-09:
Before I got your last: Added two new test directories: GIF embedded in FLAC file works but embedded TIF gives same result as above.

Caveat: TIF/TIFF is a "fey" file format (I've had previous experience from my work days). It's possible I'm creating or  embedding my TIF files with "wrong" parameters but both the image files and the track files display properly in my Linux file browser.

I'll check your new thumb-get.php after dinner.

TheOldPresbyope, 2022-08-10:
Yup, with the latest commit to thumb-gen.php, I get a 

`thumb-gen: error: imagecreatefromstring() width and/or height are 0, Path: ...`

error message thrown for TIF coverart whether embedded in FLAC track or in coverart file.

moodeaudio, 2022-08-10:
In latest commit the error messages have been streamlined and in the case where the thumbnail image can't be created the default moOde image will. be substituted.

moodeaudio, 2022-08-29:
For TIFF the cause is PHP function imagecreatefromstring() which is used in the process of creating thumbnails does not support TIFF format.

Not sure about GIF format but not worth investigating since GIF is rarely or never used for cover art IME.

TheOldPresbyope, 2022-08-29:
I agree about GIF. I tested it only because it's mentioned in coverart.php.

I have no idea how often we'll run into a user with TIFF-encoded coverart. I was inspired to test it because of the thread started by @apennc [Howdy: having problems with m4a and covers](https://moodeaudio.org/forum/showthread.php?tid=5009) just this August.

moodeaudio, 2022-08-29:
In that thread I explained to OP that TIFF is supported for displaying cover in main Playback view and lower left in Tag view (because Zend is being used) but not for thumbnails due to limitation of the PHP imagecreatefromstring() function.

I could do something similar to what is described in this post https://stackoverflow.com/questions/41775692/converting-tiff-byte-array-into-jpeg-byte-array-in-php but I think TIFF covers are rare and so I'm not inclined to add to the TODO list.

apennc, 2022-08-29:
I appreciate the followup, however my tiff’s don’t display anywhere. This an issue I can deal with by converting the tiffs to jpegs and replacing them in the meta data where all my images are located.

The bigger issue is the m4a won’t play and until I can figure that out it makes no sense to worry about cover art.

Thanks

Penn

moodeaudio, 2022-08-29:
Post a download link to one of your m4a files and I'll see if I can repro.

My test m4a alac and aac files work fine.

moodeaudio, 2022-09-04:
Will reopen if sample file is provided for testing.

closed 2022-09-04 by moodeaudio