moode-player/moode #540: Problem importing radio icons with special characters

SmartHbdev, 2022-08-30:
There is a problem when importing web radios with special characters.

If the icons have 'é' or 'è' in the name, the icon is not used / found.
The radio is still added but without the image.

Adding or editing through the interface works though.

moodeaudio, 2022-08-30:
Your best bet is to post in the support section of the Forum. I mostly use Git for developer issues.

Email Tim at moodeaudio dot org with your preferred userid and I'll set up a Forum account for you.

closed 2022-08-30 by moodeaudio

SmartHbdev, 2022-08-30:
I think this is a problem that belongs on GitHub as it is a problem related to the import code. 

Il faut que je précise peut être que j’ai crée moi même le fichier json pour importer presque 2000 radios. Il ne provient pas de la fonction export mais il est similaire.

By using the Web interface to correct manually the logo of a radio then with the export function I could see that it is necessary to use unicode with the name in the file station_data.json.
So I changed all the é by \u00e9 but with the import the code still does not find the icon.

Here is an example with the name "Générations" :

` {
            "id": 5084,
            "station": "https://generationfm.ice.infomaniak.ch/generationfm-high.mp3",
            "name": "G\u00e9n\u00e9rations",
            "type": "r",
            "logo": "local",
            "genre": "",
            "broadcaster": "",
            "language": "French",
            "country": "France",
            "region": "Europe",
            "bitrate": "",
            "format": "",
            "geo_fenced": "",
            "home_page": "http://generations.fr/",
            "reserved2": ""
        },`

As said it does not work with any of the 2 solutions: 
`"name": "G\u00e9n\u00e9rations",`

`"name": "Générations",`

The name of the icon has the name Générations.jpg

Otherwise I made 2 account requests with 2 different emails for a good half week but no answer for the moment

SmartHbdev, 2022-08-30:
I found out how to reproduce the problem.

When you import with the web interface an icon, the script copies this icon in /var/local/www/imagesw/radio-logos and at the same time makes two others in the thumb subdirectory.

When exporting the three icons, the original and the 2 thumbs are with in the zip file. Then with the import everything is included even if there are special characters in the name. Everything works in this case.

If now I delete the 2 thumbs while keeping the original, during the import the files without special characters are taken and the import script creates the 2 missing thumbs. But if the name contains special characters then we get an error like:

`ERROR: Missing station logo 'Blues Café Radio''.`

In this case the icon in /var/local/www/imagesw/radio-logos has a size of 0ko and there are no thumbs at all.
I can copy the original files in /var/local/www/imagesw/radio-logos but I would have to create the 2 thumbs myself, it's very painful when you have a lot of them.

There must be a problem in the import script so that it works with the normal names and not with the special characters. Everything else is the same, I even checked in the database, the names are the same.

It's just that the import script doesn't find the original icon when there is a special character

moodeaudio, 2022-08-31:
If you want support for this issue then make a thread in the support section of the Forum as I recommended, and post in English because I have no idea what the sentence below from your earlier post means.
```
Il faut que je précise peut être que j’ai crée moi même le fichier json pour importer presque 2000 radios. Il ne provient pas de la fonction export mais il est similaire.
```

SmartHbdev, 2022-08-31:
Oh sorry, it's a translator problem...

After more testing, the problem was with my zip file. It doesn't compress to the right format I think with a Mac.
I tested many things and applications to compress but the import always failed.
The zipped archive seemed to contain the files properly though.

The same file compressed with linux seems to import well.