moode-player/moode #545: Use systemd automounts instead of "manual" mounting for music sources

sll552, 2022-09-25:
Currently it seems that configured music sources are mounted by using the `mount` command without any entry in `/etc/fstab`.

This can be a problem if the corresponding servers go away frequently (e.g. NAS goes to sleep) or the connection is unreliable. Such manual mounts break quite easily in these situations and in the worst case leave behind a non-working mountpoint, which does not respond and slows down everything that tries to access it. 

There is however a feature of systemd called automounts. [Configuring](https://www.freedesktop.org/software/systemd/man/systemd.automount.html) is quite easy and it takes care of mounting and unmounting shares whenever needed. I expect that by using it, the stability of those mounts would be greatly improved.

moodeaudio, 2022-09-25:
Our latest r820 release has an auto-remount feature thats part of the maintenance task. It runs at 20 second intervals and looks for "Stale file handle" condition in /mnt/NAS/*. It's only been tested for an NFS mount but should work for an SMB mount as well.

sll552, 2022-09-26:
I am at 8.2.0 but the mounts did not come back, i had to remount in the settings. Thats why I manually added automounts and deleted the music sources from settings. So basically I created `/mnt/AUTO` and put my mounts there and symlinked that to the mpd library, which worked fine.

I dont want sound rude or something but in my experience those self-made solutions for this problem never _really_ work. That's also why for example kodi/xbmc doesn't mount smb volumes at all but uses some sort of smb client directly. So please consider switching to automounts which would also reduce the code required for mount handling to creating 2 units and systemctl for forced remout.

moodeaudio, 2022-09-26:
Maybe long term a replacement for the current mount system could be accomplished but it also involves developing a module for converting users existing mounts to the new method, and the backup/restore module would also need to be modified to handle the new mount method.

Given the amount of work involved unless you were willing to do the work yourself and contribute it to the project or there were a significant number of requests for something like this it's not likely to get any priority on the TODO list. You might want to post in our Forum to see if your proposal gets any traction. Email tim at moodeaudio dot org with your preferred userid and I'll create an account for you.

The other challenge is that the usage scenario you described `"servers go away frequently (e.g. NAS goes to sleep) or the connection is unreliable."` is an edge case for most users.

closed 2022-09-26 by moodeaudio

sll552, 2022-09-26:
I am currently looking into the code but unfortunately I am not very proficient with php. Just started using moode a few days ago, so its basically just stuff I needed to workaround to make it work for me.