moode-player/moode #567: show current song in browsertitle

Stephanowicz, 2022-10-21:
Hi,

may I suggest a simple extension to show the current songtitle, album... in the browsertitle?
This may be only useful for computers... the text will also show up in the taskbar

in js/playerlib.js near the metadatastuff below the cover (~ line 800), add:
```
        if (MPD.json['state'] !== 'stop') {
            document.title = MPD.json['title'] + " - " + MPD.json['album'];
        }
        else {
            document.title = "moOde Audioplayer";
        }

```
Cheers, Stephan

Stephanowicz, 2022-10-30:
If You also like to change the fav-icon to show the albumcover, You may add this in the same region like mentioned above:

`
(function(){
   $("link[rel*='icon']").each(function(){$( this ).remove();})
   var link = document.createElement('link');
   link.type = 'image/x-icon';
   link.rel = 'shortcut icon';
   link.href = $('.coverart')[0].src;
   document.getElementsByTagName('head')[0].appendChild(link);
})();
`

Cheers, Stephan

Stephanowicz, 2022-10-30:
[image]

moodeaudio, 2022-10-30:
I'm not sure of the benefit since the song title is already visible in the UI. Many users have multiple Pi's and sometimes each is opened in a Browser tab. Host name in the tab title is the natural way to easily identify them.

closed 2022-10-30 by moodeaudio

Stephanowicz, 2022-10-31:
Different people, different habits...
You see, over the years I made many suggestions for a better 'moode' & feel - but only one made it :D 
Ok, so be it - but I really I would like to strongly suggest that you think about an implementation of modules  and to develop a concept that makes it easier to add features like this. 
Maybe there are people who.... but there are many others who don't

Cheers, Stephan

moodeaudio, 2022-10-31:
I can assure you its nothing personal. Our project gets many requests for UI changes and only a tiny fraction are adopted because most of the requests boil down to individual preference and don't end up having widespread appeal.

I'm not sure what you mean by "modules" but there is a set of developer guides in the `docs` repo that have instructions for setting up the build environment for moOde. We use standard tooling for example Gulp for building the front-end, Debian packaging for all the components and pi-gen to make the images. 
https://github.com/moode-player/docs

Stephanowicz, 2022-10-31:
With 'modules' I meant a more modular approach. At moment it's very complex to add a new feature which is due to the current structure of the code. You probably won't get around to completly revise the code sooner or later...

Regarding the 'individual preferences' - well, I don't claim these ideas to be mine - I've seen them somewhere else before - so there must be at least another person with this 'personal preference' :D 
I assume the high inhibition threshold is due to the quite complex process of implementing and configuring those features - which is sad, as they really would enrich the user experience and feel.  
Well, my point of view 

Cheers, Stephan