moode-player/moode #572: Bluetooth intermitance issue

Zageron, 2022-10-29:
This problem started to arise after I upgraded to a Topping D10 Balanced from a Schiit Modi.
```
Oct 28 23:01:11 moode bluealsa[122466]: /usr/bin/bluealsa: D: ba-transport.c:928: Closing BT: 10
Oct 28 23:01:11 moode bluealsa[122466]: /usr/bin/bluealsa: D: ba-transport.c:1054: Exiting IO thread: A2DP Sink (SBC)
```

Reproduction:
- Connect to Bluetooth
- Use bluetooth (60 seconds on phone, a few minutes on laptop)
- Audio cuts out.

Side effects:
- Bluetooth remains connected but does not output any audio.
- Disconnecting and reconnecting does not result in audio stream.
- Disconnecting, Restarting bluetooth controller, then re connecting is the only way to get the audio back.

Another example (this actually disconnected):
```
Oct 28 23:15:49 moode bluealsa[138324]: /usr/bin/bluealsa: D: bluez.c:332: Disconnecting media endpoint: /org/bluez/hci0/A2DP/SBC/Si>
Oct 28 23:15:49 moode bluealsa[138324]: /usr/bin/bluealsa: D: ba-transport.c:901: Releasing A2DP transport: A2DP Sink (SBC)
Oct 28 23:15:49 moode bluealsa[138324]: /usr/bin/bluealsa: D: ba-transport.c:928: Closing BT: 10
Oct 28 23:15:49 moode bluealsa[138324]: /usr/bin/bluealsa: D: ba-transport.c:1054: Exiting IO thread: A2DP Sink (SBC)
Oct 28 23:15:49 moode bluealsa[138324]: /usr/bin/bluealsa: D: ba-transport.c:1006: Closing PCM: 11
Oct 28 23:15:49 moode bluealsa[138324]: /usr/bin/bluealsa: D: bluez.c:1009: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRem>
Oct 28 23:15:49 moode bluealsa[138324]: /usr/bin/bluealsa: D: bluez.c:1009: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRem>
Oct 28 23:15:49 moode bluealsa[138324]: /usr/bin/bluealsa: D: bluez.c:1009: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRem>
Oct 28 23:15:49 moode bluealsa[138324]: /usr/bin/bluealsa: D: bluez.c:1009: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRem>
Oct 28 23:15:49 moode bluealsa[138324]: /usr/bin/bluealsa: D: ba-transport.c:361: Freeing transport: A2DP Sink (SBC)
Oct 28 23:15:49 moode bluealsa[138324]: /usr/bin/bluealsa: D: ba-device.c:131: Freeing device: C4:BD:E5:1B:8D:BF
```

moodeaudio, 2022-10-29:
If all that was done was to change the USB audio device then it would suggest the issue is with the device or the interconnect. Your best bet is to post the issue in the Forum in case someone else has that same device or has a similar issue. 

Email tim at moodeaudio dot org with your preferred userid and I'll setup an account for you.

closed 2022-10-29 by moodeaudio

Zageron, 2022-10-29:
The Spotify renderer works flawlessly, continuously, with no issue.
Sent an email.