moode-player/moode #576: "pointer: fine" too restrictive?

Stephanowicz, 2022-11-01:
Hi,

in moode.css:
```
/* Use custom scroll bars for desktop monitors */
@media (pointer: fine){
	.custom-scrollbars ::-webkit-scrollbar {
		width: 11px;
	}
 ...
```
my tablet doesn't hit this query, so I don't get the nicer small scroll-bars
[image]

Changing it to
```
/* Use custom scroll bars for desktop monitors */
@media (any-pointer: fine),(any-pointer: coarse) {
	.custom-scrollbars ::-webkit-scrollbar {
		width: 11px;
	}
gives me this result
````
[image]

I don't know why the api thinks I have a coarse pointer... nevertheless I have no problems using touch, touchpad or my mouse to scroll the lists with the tiny scroll-bars...

Question: 
any reason for the 'fine pointer'  restriction? 

Cheers, Stephan

moodeaudio, 2022-11-01:
I'm not sure. It was another dev that added that bit of code way back when.

I'll read up on it and see what makes sense. What model Tablet do you have?

Stephanowicz, 2022-11-01:
I found an entry  in bugzilla - they seemed to have fixed it, so it's recognized properly and working with `any-pointer: fine`  in firefox.
(https://bugzilla.mozilla.org/show_bug.cgi?id=1638556)
I posted the bug @ MS tech community - let's see what they will do...

So in summary one can say that 
a) from the specs one has to use `any-pointer` for media queries, in order to make it work for tablets with touch device having also a touchpad or mouse, as the 'coarse' touch display will always be the first device that is recognized by the api.
b) chromium and maybe others have a bug in their api that prevents `any-pointer:fine` getting fired when a mouse or touchpad is attached to a device touch display.

Here's a link for testing the browsers reponses for media queries https://patrickhlauke.github.io/touch/pointer-hover-any-pointer-any-hover/

Cheers, Stephan

moodeaudio, 2022-11-01:
Does this mean not change needed in the CSS?

Stephanowicz, 2022-11-02:
Well, the question is: is it necessary and useful to distinguish between coarse and fine pointers regarding the scrollbar?
If so, the css has to be changed from `pointer: fine` to `any-pointer: fine` 
If not, I would remove it.

@moment any touch-devices will be recognized as 'coarse' even if the have a mouse attached or a touchpad.

moodeaudio, 2022-11-02:
Sure I can change it `any-pointer: fine`.

After reading the bug report it looks like its just a matter of waiting for the Browser devs to fix the bugs.

Stephanowicz, 2022-11-02:
sorry, You should close when done :D

moodeaudio, 2022-11-02:
Committed.

closed 2022-11-02 by moodeaudio