moode-player/moode #587: scripts-library - querying empty array

Stephanowicz, 2022-12-10:
Hi,

just stumbled across this: 
in scripts-library ~lines 846-850:

`    // Display album name heading:
    // - if more than 1 album for clicked artist
    // - if first song musicbrainz_albumid != '0'
	if ((filteredAlbums.length > 1 && LIB.artistClicked == true && LIB.albumClicked == false) ||
        filteredSongs[0].mb_albumid != '0') {
		$('.lib-album-heading').css('display', 'block');
	}
`
Array `filteredSongs` is not checked wether empty - in my case it is empty and throws an error.
Suggestion: add query  `filteredSongs.length > 0`

Cheers, Stephan

moodeaudio, 2022-12-10:
Is there something about the albums that causes filteredSongs to be empty?

Stephanowicz, 2022-12-10:
Hi,
yes, this could be the case - I had to rebuild the lib.

moodeaudio, 2022-12-10:
First I'd need to repro the issue in order to understand how it could happen. I don't recall any similar issues in the Forum so maybe it's just something unique to your particular config and collection. Maybe some bad tag data in a track. Something like that.

Stephanowicz, 2022-12-10:
Ok, I just wanted to mention this - I had no problems with using the interface, I just stumbled over the error message when opening the debug console.

Cheers,  Stephan

moodeaudio, 2022-12-12:
Will reopen if similar issues are reported in the Forum.

closed 2022-12-12 by moodeaudio