moode-player/moode #588: Any chance to get support for mpdscribble?

Ghoost666, 2022-12-16:
Hello, 
i think i asked this in the official forum some time ago but it was denied (i think it was because the used scrobble software broke some time ago and development of it halted and there was a lack of good alternatives)

but i just saw that there is a official scrobble software from MPD https://github.com/MusicPlayerDaemon/mpdscribble which also seems to be actively development and there are also debian packages, any chance to get support for it in moode again?

moodeaudio, 2022-12-16:
It could be added if there is more demand for something like this but since the original, unmaintained mpdas scrobbler was removed in 2020 as part of the moOde 7.0.0 release there have been maybe 1 or 2 requests and so it's not really on my radar.

There is always a possibility that another dev will want to do the work to integrate it though.

Ghoost666, 2022-12-17:
thanks for the response

i think its a fairly easy program to run aditionally since it just runs in the background and probably plays well together with MPD
i could test it (since i probably will add it myself anyway if you guys dont decide to integrate it natively)
i dont have "special" need for it (im not a big last.fm user so far but playing stats and new song discovery based of it would be interesting to try) but overall i just thought its a nice and easy addition
 
but i have no idea how to integrate it into the webinterface since im not dev, maybe someone else can help here

i can report back once i tried it through the command line

closed 2022-12-17 by moodeaudio