moode-player/moode #589: moOde v8.2.3 64bit image not booting on RPi 4B

ek9, 2022-12-20:
Latest moOde v8.2.3 64bit image is not booting on Raspberry Pi 4B.

Using image `2022-12-18-moode-r823-arm64-lite.img.zip` and writing it with `rpi-imager`

The boot procedure hangs with error:

```
[ 1.42XXXXXX] hub 4-0:1.0: config failed, hub doesn't have any ports! (err: -19)
```

I'm assuming this is kernel error
Using an older image file boots and works fine (i.e. `2022-10-24-moode-r822-arm64-lite.`).

moodeaudio, 2022-12-20:
Works fine on my end
```
Authenticated to moode ([192.168.1.136]:22).
Linux: 5.15.76-v8+ #1597 | aarch64 (64-bit) | DebOS: 11.6
Model: Pi-4B 1.1 1GB
Audio: Allo Boss DAC

                      ____     __      ___
          __ _  ___  / __ \___/ /__   ( _ )
         /  ' \/ _ \/ /_/ / _  / -_) / _  |
        /_/_/_/\___/\____/\_,_/\__/  \___/

                 moOde audio player
              Release 8.2.3 2022-12-18
                (C) 2014 Tim Curtis

The programs included with moOde are free software; the
exact distribution terms for each program are described
in the individual files in /usr/share/doc/*/copyright or
in the moOde source files. moOde comes with absolutely
no warranties either expressed or implied, or any other
such guarantees.

To set a new password for the user 'pi' type 'passwd'.

Last login: Tue Dec 20 10:28:14 2022 from 192.168.1.125
pi@moode:~ $ 
```

ek9, 2023-01-03:
Redownloaded newer (8.2.4) image - seems to be working fine. Closing.

closed 2023-01-03 by ek9