moode-player/moode #590: 8.2.3 Airplay receiver no longer visible

hexive, 2022-12-23:
A change between 8.2.2 and 8.2.3 makes Moode no longer appear as airplay receiver.

I tested as follows on a Pi4 B 1.5

With a clean 8.2.2 card from image_2022-10-24-moode-r822-arm64-lite.img
- boot; add wifi credentials; restart
- turn on Airplay; click set
- Moode Airplay visible

With a clean 8.2.3 card from image_2022-12-18-moode-r823-arm64-lite.img
- boot; add wifi credentials; restart
- turn on Airplay; click set
- Moode Airplay not visible

Any suggestions on how to troubleshoot this? Thanks in advance.

moodeaudio, 2022-12-23:
Yes, AirPlay start is broken on the r823 image. This Forum thread explains the issue https://moodeaudio.org/forum/showthread.php?tid=5256&pid=44304#pid44304

Here is a patch to fix it 
```
sudo sed -i "/audio_backend_buffer_desired_length_in_seconds'/d" /etc/shairport-sync.conf
```

Your best bet for ongoing support is to get a Forum account cos I mainly use Git for developer issues. Email tim at moodeaudio dot org with your preferred userid and I'll create an account for u.

There will be an 8.2.4 bugfix release sometime b4 the end of the year.

hexive, 2022-12-23:
Ah! Many thanks and apologies for not looking in the forum first.

closed 2022-12-23 by hexive