moode-player/moode #596: Gapless playback fails when resampling in mpd

charleski, 2023-01-18:
When upsampling audio, gapless playback fails as the following track skips the start of the track.

Max Kellerman (mpd dev) refuses to look at this since moode uses a patch to enable selective resampling. 

[image]

mpd log:
Jan 18 11:15 : client: [1] process command list returned 0
Jan 18 11:15 : client: [1] closed
Jan 18 11:15 : client: [2] opened from 127.0.0.1:59622
Jan 18 11:15 : client: [2] process command "status"
Jan 18 11:15 : client: [2] command returned 0
Jan 18 11:15 : client: [2] process command "currentsong"
Jan 18 11:15 : client: [2] command returned 0
Jan 18 11:15 : client: [2] process command "lsinfo "USB/Music_Ext/Unfiled/Rinaldo test/15 Scena 5 - Furie terribili (Aria - Armida)-frgm.flac""
Jan 18 11:15 : client: [2] command returned 0
Jan 18 11:15 : client: [2] process command "close"
Jan 18 11:15 : client: [2] command returned 4
Jan 18 11:15 : client: [2] closed
Jan 18 11:15 : client: [3] opened from 127.0.0.1:59624
Jan 18 11:15 : client: [3] process command "status"
Jan 18 11:15 : client: [3] command returned 0
Jan 18 11:15 : client: [3] process command "idle"
Jan 18 11:15 : client: [3] command returned 1
Jan 18 11:15 : client: [4] opened from 127.0.0.1:59626
Jan 18 11:15 : client: [4] process command "currentsong"
Jan 18 11:15 : client: [4] command returned 0
Jan 18 11:15 : client: [4] process command "close"
Jan 18 11:15 : client: [4] command returned 4
Jan 18 11:15 : client: [4] closed
Jan 18 11:15 : client: [5] opened from 127.0.0.1:59628
Jan 18 11:15 : client: [6] opened from 127.0.0.1:59632
Jan 18 11:15 : client: [6] process command "currentsong"
Jan 18 11:15 : client: [6] command returned 0
Jan 18 11:15 : client: [6] process command "close"
Jan 18 11:15 : client: [6] command returned 4
Jan 18 11:15 : client: [6] closed
Jan 18 11:15 : client: [7] opened from 127.0.0.1:59634
Jan 18 11:15 : client: [7] process command "lsinfo"
Jan 18 11:15 : client: [7] command returned 0
Jan 18 11:15 : client: [7] process command "close"
Jan 18 11:15 : client: [7] command returned 4
Jan 18 11:15 : client: [7] closed
Jan 18 11:15 : client: [8] opened from 127.0.0.1:59636
Jan 18 11:15 : client: [8] process command "status"
Jan 18 11:15 : client: [8] command returned 0
Jan 18 11:15 : client: [8] process command "currentsong"
Jan 18 11:15 : client: [8] command returned 0
Jan 18 11:15 : client: [5] process command "close"
Jan 18 11:15 : client: [5] command returned 4
Jan 18 11:15 : client: [5] closed
Jan 18 11:15 : client: [8] process command "lsinfo "USB/Music_Ext/Unfiled/Rinaldo test/15 Scena 5 - Furie terribili (Aria - Armida)-frgm.flac""
Jan 18 11:15 : client: [8] command returned 0
Jan 18 11:15 : client: [8] process command "close"
Jan 18 11:15 : client: [8] command returned 4
Jan 18 11:15 : client: [8] closed
Jan 18 11:15 : client: [9] opened from 127.0.0.1:59638
Jan 18 11:15 : client: [9] process command "status"
Jan 18 11:15 : client: [9] command returned 0
Jan 18 11:15 : client: [9] process command "idle"
Jan 18 11:15 : client: [9] command returned 1
Jan 18 11:15 : client: [10] opened from 127.0.0.1:59640
Jan 18 11:15 : client: [10] process command "playlistinfo"
Jan 18 11:15 : client: [10] command returned 0
Jan 18 11:15 : client: [10] process command "close"
Jan 18 11:15 : client: [10] command returned 4
Jan 18 11:15 : client: [10] closed
Jan 18 11:15 : client: [11] opened from 127.0.0.1:59642
Jan 18 11:15 : client: [11] process command "currentsong"
Jan 18 11:15 : client: [11] command returned 0
[... 269 lines cut ...]
Tag plugins:
id3tag

Output plugins:
shout null fifo pipe alsa httpd recorder

Encoder plugins:
null vorbis opus lame wave flac

Archive plugins:
[bz2] bz2
[zzip] zip
[iso] iso

Input plugins:
file io_uring archive alsa curl ffmpeg cdio_paranoia

Playlist plugins:
extm3u m3u pls xspf asx rss flac cue embcue

Protocols:
file:// alsa:// cdda:// ftp:// ftps:// gopher:// hls+http:// hls+https:// http:// https:// mmsh:// mmst:// rtmp:// rtmps:// rtmpt:// rtmpts:// rtp:// rtsp:// rtsps:// scp:// sftp:// smb:// srtp://

Other features:
avahi dbus epoll icu inotify systemd tcp un

moodeaudio, 2023-01-18:
Our "Selective Resampling" patch doesn't alter how the SoX resampler itself works it only serves to decide whether to engage the resampler before track play based on the track's sample rate. 

It's possible of course that this logic somehow causes the artifact you are seeing in the waveform at the transition between gapless tracks.

1. Does the issue occur only when Selective Resampling is enabled?
2. What software is being used to render the waveforms

charleski, 2023-01-18:
Yes, I can't see how your patch would cause this behaviour, but he's using it as a reason to label moode as 'unsupported'.
https://github.com/MusicPlayerDaemon/MPD/issues/1692

This example was captured with the resampler set to 192kHz without Selective Reasmpling.
The waveforms were captured using arecord with the loopback turned on. arecord captured the resampled waveform at 192kHz and the unresampled wave at 44.1kHz (original is 44.1kHz)
The image shown is from Audition. I used Audition to resample the 44.1kHz wave to 192kHz and then inserted both into a multitrack session and aligned them manually.

bitkeeper, 2023-02-07:
@charleski I also think also that the patch has nothing to do with it, but to take it out of the picture you can upgrade/downgrade to the latest version from debian.

moodeaudio, 2023-03-12:
Below is a recipe that can be used to create a generic MPD test platform from stock RaspiOS and MPD sources. It can be used to try and reproduce the issue. The results can then be submitted to MPD maintainer.

I'll reopen this issue if there is some activity toward the generic test.

```
#
# Build MPD from sources
#

# 
# Use Raspberrry Pi Imager
# - Select "Raspberry Pi OS Lite (64-bit)"
# - Click the gear icon: Enable SSH, set Username pi and password then SAVE
# - Write the image
#

#
# After first boot
# - ssh pi@raspberrypi
# - sudo apt update
# - sudo apt -y upgrade
# - sudo reboot
#

#
# Install core packages and build tooling
#

sudo apt -y install ffmpeg mpc
sudo apt -y install cmake git clang-11 meson

#
# Create MPD runtime environment
#

sudo useradd mpd
sudo mkdir /var/lib/mpd
sudo mkdir /var/lib/mpd/music
sudo mkdir /var/lib/mpd/playlists
sudo touch /var/lib/mpd/state
sudo chown -R mpd:audio /var/lib/mpd
sudo mkdir /var/log/mpd
sudo touch /var/log/mpd/log
sudo chmod 644 /var/log/mpd/log
sudo chown -R mpd:audio /var/log/mpd
sudo touch /etc/mpd.conf
sudo chown mpd:audio /etc/mpd.conf
sudo chmod 0666 /etc/mpd.conf

#
# Install dev libs
#

sudo apt -y install \
libyajl-dev \
libasound2-dev \
libavahi-client-dev \
libboost-all-dev \
libavcodec-dev \
libavformat-dev \
libbz2-dev \
libcdio-paranoia-dev \
libcurl4-gnutls-dev \
libfaad-dev \
libflac-dev \
libglib2.0-dev \
libicu-dev \
libid3tag0-dev \
libiso9660-dev \
libmad0-dev \
libmpdclient-dev \
libmpg123-dev \
libmp3lame-dev \
libshout3-dev \
libsoxr-dev \
libsystemd-dev \
libvorbis-dev \
libwavpack-dev \
libwrap0-dev \
libzzip-dev \
libpcre++-dev

#
# Download MPD sources
#

BRANCH=0.23
VERSION=.12
sudo wget http://www.musicpd.org/download/mpd/$BRANCH/mpd-$BRANCH$VERSION.tar.xz
sudo tar xf mpd-$BRANCH$VERSION.tar.xz
cd mpd-$BRANCH$VERSION
sudo meson . output/release --buildtype=release -Db_ndebug=true

#
# Configure compile options
#

sudo meson configure \
-Dalsa=enabled \
-Dbzip2=enabled \
-Dcurl=enabled \
-Ddatabase=true \
-Ddsd=true \
-Dffmpeg=enabled \
-Dfaad=enabled \
-Dflac=enabled \
-Dhttpd=true \
-Did3tag=enabled \
-Dlame=enabled \
-Dlibmpdclient=enabled \
-Dmad=enabled \
-Dmpg123=enabled \
-Dpipe=true \
-Drecorder=true \
-Dshout=enabled \
-Dsoxr=enabled \
-Dvorbis=enabled \
-Dwave_encoder=true \
-Dwavpack=enabled \
-Dzeroconf=avahi \
-Dpcre=enabled \
-Dzzip=enabled \
-Dao=disabled \
-Daudiofile=disabled \
-Ddbus=disabled \
-Dexpat=disabled \
-Dfluidsynth=disabled \
-Dgme=disabled \
-Dipv6=disabled \
-Djack=disabled \
-Dlibsamplerate=disabled \
-Dnfs=disabled \
-Doss=disabled \
-Dpulse=disabled \
-Dsidplay=disabled \
-Dsmbclient=disabled \
-Dsndfile=disabled \
-Dsqlite=disabled \
-Dudisks=disabled \
-Dupnp=disabled \
-Dwildmidi=disabled \
-Dsoundcloud=disabled \
-Dtidal=disabled \
-Dqobuz=disabled
output/release

#
# Compile and install MPD
#

sudo ninja -C output/release
sudo ninja -C output/release install

# 
# Add location of mpd.conf to ExecStart
#
sudo nano /usr/local/lib/systemd/system/mpd.service
ExecStart=/usr/local/bin/mpd --systemd /etc/mpd.conf

#
# Paste below into /etc/mpd.conf
# Change plughw:0,0 and hw:0 depending on which card has Headphones assigned.
# For example if its assigned to card 1 then plughw:1,0 and hw:1
#
sudo nano /etc/mpd.conf

music_directory "/var/lib/mpd/music"
playlist_directory "/var/lib/mpd/playlists"
db_file "/var/lib/mpd/database"
log_file "/var/log/mpd/log"
pid_file "/var/run/mpd/pid"
state_file "/var/lib/mpd/state"
sticker_file "/var/lib/mpd/sticker.sql"
user "mpd"
group "audio"
bind_to_address "any"
port "6600"
log_level "default"
restore_paused "yes"
auto_update "no"
follow_outside_symlinks "yes"
follow_inside_symlinks "yes"
zeroconf_enabled "no"
zeroconf_name "Moode MPD"
filesystem_charset "UTF-8"
metadata_to_use "+comment"
replaygain "off"
replaygain_preamp "0"
volume_normalization "no"
audio_buffer_size "4096"
max_output_buffer_size "131072"
max_playlist_length "16384"
max_connections "128"

decoder {
plugin "ffmpeg"
enabled "yes"
}

input {
plugin "curl"
}

resampler {
plugin "soxr"
quality "high"
threads "1"
}

audio_output {
type "alsa"
name "ALSA Default"
device "plughw:0,0"
mixer_type "hardware"
mixer_control "Headphone"
mixer_device "hw:0"
mixer_index "0"
dop "no"
stop_dsd_silence "no"
thesycon_dsd_workaround "no"
}

#
# Verify working installation
#

sudo systemctl start mpd
mpc
mpc volume 10

#
# End
#
```

closed 2023-03-12 by moodeaudio