moode-player/moode #603: Favorites playlist deleted on in-place update

amanita-citrina, 2023-02-09:
During in-place updates, my "Favorites" playlist at `/var/lib/mpd/playlists/Favorites.m3u` is replaced by the empty default `Favorites.m3u`. Possibly, the default playlist is treated in the same way, but I never cared about that one.

Recently, this has happened when updating from 8.2.x to 8.2.5.

**Note:**<br />I have a special "dual-boot" setup with multiple partitions that enables major upgrades on the same SD card. Otherwise, I would have to open the enclosure of the player to switch SD cards, which is no fun at all.

I don't think that this has anything to do with it, but who knows...

moodeaudio, 2023-02-09:
I'll add to the TODO list to investigate.

amanita-citrina, 2023-03-23:
After updating to 8.3.0 I checked again and found that only the Favorites playlist is overwritten. The default playlist remains untouched. I have no other playlists.

moodeaudio, 2023-03-23:
It appears to be a bug and is on the TODO list for upcoming 8.3.1 release.

moodeaudio, 2023-03-29:
Fixed for upcoming 8.3.1 release (Q2 2023)

closed 2023-03-29 by moodeaudio