moode-player/moode #605: issues after usb device gets disconect

tomaxsas, 2023-02-19:
Using usb dac. When dac is turned of mpd stops playing radio. After turning dac on, and press play, radio is getting played from cache, and stops after few seconds. If hit play then pause then play again current stream plays. Not sure, this might be mpd bug not moode...

Another thing with spotify, if dac is turned off while playing spotify, need to restart spotify renderer cause it does not show up

moodeaudio, 2023-02-26:
It might take some time before Linux USB and ALSA pickup the DAC after its turned on.

Spotify Connect currently does not handle loss of the audio connection and it hangs. I think its on their TODO list to address this in a future release.

tomaxsas, 2023-02-26:
reconnect time is instant, i'm bothered about radio issue :D. but this must be mpd issue, just need to test it with vanilla mpd, cause saw they say moode mpd is altered

moodeaudio, 2023-02-26:
The radio issue does not sound like an MPD bug per se and our patch to MPD only serves to provide the super nice Selective Resampling feature and has nothing to do with other parts of MPD. The MPD maintainer is correct in denying support for any type of patched MPD. Our project like most does same for modified releases.

If you really want to determine whether this could be an MPD bug then I'd recommend doing a vanilla OS and MPD configuration. That way you rule out any possible configuration issue in moOde itself.

Here's a recipe that I've used in the past.
```
#
# Build MPD from sources
#

# 
# Use Raspberrry Pi Imager
# - Select "Raspberry Pi OS Lite (64-bit)"
# - Click the gear icon: Enable SSH, set Username and password then SAVE
# - Write the image
#

#
# After first boot
# - ssh pi@raspberrypi
# - sudo apt update
# - sudo apt -y upgrade
# - sudo reboot
#

#
# Install core packages and build tooling
#

sudo apt -y install ffmpeg mpc
sudo apt -y install cmake git clang-11 meson

#
# Create MPD runtime environment
#

sudo useradd mpd
sudo mkdir /var/lib/mpd
sudo mkdir /var/lib/mpd/music
sudo mkdir /var/lib/mpd/playlists
sudo touch /var/lib/mpd/state
sudo chown -R mpd:audio /var/lib/mpd
sudo mkdir /var/log/mpd
sudo touch /var/log/mpd/log
sudo chmod 644 /var/log/mpd/log
sudo chown -R mpd:audio /var/log/mpd
sudo touch /etc/mpd.conf
sudo chown mpd:audio /etc/mpd.conf
sudo chmod 0666 /etc/mpd.conf

#
# Install dev libs
#

sudo apt -y install \
libyajl-dev \
libasound2-dev \
libavahi-client-dev \
libboost-all-dev \
libavcodec-dev \
libavformat-dev \
libbz2-dev \
libcdio-paranoia-dev \
libcurl4-gnutls-dev \
libfaad-dev \
libflac-dev \
libglib2.0-dev \
libicu-dev \
libid3tag0-dev \
libiso9660-dev \
libmad0-dev \
libmpdclient-dev \
libmpg123-dev \
libmp3lame-dev \
libshout3-dev \
libsoxr-dev \
libsystemd-dev \
libvorbis-dev \
libwavpack-dev \
libwrap0-dev \
libzzip-dev \
libpcre++-dev

#
# Download MPD sources
#

BRANCH=0.23
VERSION=.12
sudo wget http://www.musicpd.org/download/mpd/$BRANCH/mpd-$BRANCH$VERSION.tar.xz
sudo tar xf mpd-$BRANCH$VERSION.tar.xz
cd mpd-$BRANCH$VERSION
sudo meson . output/release --buildtype=release -Db_ndebug=true

#
# Configure compile options
#

sudo meson configure \
-Dalsa=enabled \
-Dbzip2=enabled \
-Dcurl=enabled \
-Ddatabase=true \
-Ddsd=true \
-Dffmpeg=enabled \
-Dfaad=enabled \
-Dflac=enabled \
-Dhttpd=true \
-Did3tag=enabled \
-Dlame=enabled \
-Dlibmpdclient=enabled \
-Dmad=enabled \
-Dmpg123=enabled \
-Dpipe=true \
-Drecorder=true \
-Dshout=enabled \
-Dsoxr=enabled \
-Dvorbis=enabled \
-Dwave_encoder=true \
-Dwavpack=enabled \
-Dzeroconf=avahi \
-Dpcre=enabled \
-Dzzip=enabled \
-Dao=disabled \
-Daudiofile=disabled \
-Ddbus=disabled \
-Dexpat=disabled \
-Dfluidsynth=disabled \
-Dgme=disabled \
-Dipv6=disabled \
-Djack=disabled \
-Dlibsamplerate=disabled \
-Dnfs=disabled \
-Doss=disabled \
-Dpulse=disabled \
-Dsidplay=disabled \
-Dsmbclient=disabled \
-Dsndfile=disabled \
-Dsqlite=disabled \
-Dudisks=disabled \
-Dupnp=disabled \
-Dwildmidi=disabled \
-Dsoundcloud=disabled \
-Dtidal=disabled \
-Dqobuz=disabled
output/release

#
# Compile and install MPD
#

sudo ninja -C output/release
sudo ninja -C output/release install

# 
# Add location of mpd.conf to ExecStart
#
sudo nano /usr/local/lib/systemd/system/mpd.service
ExecStart=/usr/local/bin/mpd --systemd /etc/mpd.conf

#
# Paste below into /etc/mpd.conf
# Change plughw:0,0 and hw:0 depending on which card has Headphones assigned.
# For example if its assigned to card 1 then plughw:1,0 and hw:1
#
sudo nano /etc/mpd.conf

music_directory "/var/lib/mpd/music"
playlist_directory "/var/lib/mpd/playlists"
db_file "/var/lib/mpd/database"
log_file "/var/log/mpd/log"
pid_file "/var/run/mpd/pid"
state_file "/var/lib/mpd/state"
sticker_file "/var/lib/mpd/sticker.sql"
user "mpd"
group "audio"
bind_to_address "any"
port "6600"
log_level "default"
restore_paused "yes"
auto_update "no"
follow_outside_symlinks "yes"
follow_inside_symlinks "yes"
zeroconf_enabled "no"
zeroconf_name "Moode MPD"
filesystem_charset "UTF-8"
metadata_to_use "+comment"
replaygain "off"
replaygain_preamp "0"
volume_normalization "no"
audio_buffer_size "4096"
max_output_buffer_size "131072"
max_playlist_length "16384"
max_connections "128"

decoder {
plugin "ffmpeg"
enabled "yes"
}

input {
plugin "curl"
}

resampler {
plugin "soxr"
quality "high"
threads "1"
}

audio_output {
type "alsa"
name "ALSA Default"
device "plughw:0,0"
mixer_type "hardware"
mixer_control "Headphone"
mixer_device "hw:0"
mixer_index "0"
dop "no"
stop_dsd_silence "no"
thesycon_dsd_workaround "no"
}

#
# Verify working installation
#

sudo systemctl start mpd
mpc
mpc volume 10

#
# End
#
```

moodeaudio, 2023-03-12:
User @meszarosa posted a vanilla repro in this MD issue
https://github.com/MusicPlayerDaemon/MPD/issues/1579

closed 2023-03-12 by moodeaudio