moode-player/moode #606: upmpdcli ignores "foreign" network (ip multicast routing and upnp) and connection fails

cilahn, 2023-02-26:
Hello,
when using multicast routing on my home network request from the **not** local attached network gets ignored. Hiding the foreign network behind nat fixes this issue. I digged into the upmdpcli.conf options if there is any restriction that non local networks get declined, but it seems hardcoded or handled by moode somewhere else.

When running a upnp controler on the same network as moode (i.e. Bubble PNP or kodi). The SSDP announcment and following request for the desc.xml via http work fine.

The issue was seen in the following setup:

`192.168.80.243` (jellyfin server) --> firewall with pim-sm router --> `10.10.10.66` (moode player with upnp renderer enabled)

upmpdcli is sending to port 1900 multicast SSDP Packets and announces the listening port 8200 and 49152 for incoming request for the Upnp description xml.

Sniffing on the jellyfin server shows incoming SSDP announcements from moode:
`01:18:00.557750 IP 10.10.10.66.39206 > 239.255.255.250.1900: UDP, length 478

inside the packet the url: 
`http://10.10.10.66:49152/uuid-5e211097-4658-5161-a661-801f028f23ff/description.xml
`is announced.

Jellyfin tries to reach out to this url to get more information about the renderer, but this fails. The jellyfin log show a connection reset:

```
[2023-02-26 01:26:45.106 +01:00] [DBG] Attempting to create PlayToController from location http://10.10.10.66:49152/uuid-5e211097-4658-5161-a661-801f028f23ff/description.xml
[2023-02-26 01:26:45.146 +01:00] [ERR] Error creating PlayTo device.
System.Net.Http.HttpRequestException: An error occurred while sending the request.
 ---> System.IO.IOException: Unable to read data from the transport connection: Connection reset by peer.
 ---> System.Net.Sockets.SocketException (104): Connection reset by peer

```
On the moode side i can sniff the incoming request, but no answer is sent out
After applying a nat rule on the firewall in between the two networks mapping the source ip 192.168.80.243 to the firewall interface 10.10.10.2 on the same network as moode the connection is handled, the xml file delivered and the renderer shown in the jellyfin menu for remote devices. Playback than works like a charm.

So can this be fixed?

Thanks for moode, despite the bug i had a lot of fun with it!

moodeaudio, 2023-02-26:
I'm not sure what you mean by `"when using multicast routing on my home network"` and `"foreign network"`. The typical home network is a single LAN / WLAN address space.

cilahn, 2023-02-26:
My setup is not typical, my home network is divided into two networks (ethernet 192.168.80.0/24 and wireless 10.10.10.0/24) both are connected via a firewall/router. The 'foreign' is a bit unlucky and can be replaced by the 192.168.80.0/24 internal network. Hope this clears it a bit.

moodeaudio, 2023-02-26:
Right but why do you think this is a moOde issue? First of all we just use unmodified upmpdci and Linux standard networking, secondly forwarding Multicast and Broadcast packets between two different IP address spaces has to be explicitly configured in the routing device.

closed 2023-02-26 by moodeaudio

cilahn, 2023-02-26:
So you can confirm that moode is not blocking any traffic going to or from upmpdcli in anyway? 
I started here first to be sure it's only an upmpdcli issue and not moode related. I also thought you where interested in this because connection two multicast domains isn't that unusual.

moodeaudio, 2023-02-26:
Right, there is no firewall or manipulation of IP traffic in normal configuration. It's just the standard Linux network stack Have a look at the Network section in the repo file `www/daemon/worker.php` and the file `www/inc/network.php` to see how the configuration is created.

A firewall is created only if Router mode is enabled in Network Config UI but this only applies when the Access Point starts up. This is a private WLAN on wlan0 interface that uses 172.24.1/24 address space. The iftable rule set is only designed for forwarding packets from wlan0 172 to the eth0 interface.