moode-player/moode #611: moOde players aren't discovered by MS Windows hosts

TheOldPresbyope, 2023-03-25:
Numerous moOde users have posted to the forum that their moOde player doesn't appear among the hosts displayed in their MS Windows Network panel. 

Both the cause and a possible solution are described in the README.md file at christgau's [wsdd repo](https://github.com/christgau/wsdd). This repo contains the single executable, the wsdd.py daemon.

In my tests, starting up the wsdd.py daemon as user "pi" from the command line in my moOde 8.3.0 player is sufficient for this player to appear in the Network panel on my Win10Pro (guest) host. Assuming the moOde player has been configured to enable SMB file sharing, all its shares can be displayed and used via the Windows GUI. 

The README claims wsdd is slated for inclusion in the next Debian release (Bookworm) and it is already included in other Linux distros, so it seems a safe bet to include it in moOde. 

I believe wsdd is a good candidate for inclusion in the next moOde update, perhaps to be enabled/disabled via the System Config panel. This involves three files: the Python executable, the man page, and the service file. (Mind the details in the sample service file.) We could wait for Raspberry Pi OS to bump to Bookworm but that seems unnecessary.

-----
Note that it has always been possible simply to map moOde shares to Windows network drives either in the GUI Network panel or via the "net use" command but it seems this is too tech-y a solution for a number of users.

dheijl, 2023-03-25:
If the UPNP/DLNA renderer is enabled it is visible. Maybe worth documenting?

[image]

TheOldPresbyope, 2023-03-26:
While this is true, I'm not sure that it needs to be documented somewhere. It exposes the existence of the UPnP Renderer (if enabled) but doesn't provide any functionality on its own. Clicking on it results in the upmpdcli presentation file being displayed in the Windows web browser.

At present this is just the default file, which begins "Hi, this is the default upmpdcli presentation page and it is not very interesting." I suppose we could change this to be slightly more interesting but it still isn't very useful. One needs a UPnP Control Point.

dheijl, 2023-03-26:
OK, just thought I would mention it.

moodeaudio, 2023-03-30:
I could add the python script and service file but looking at the issue list in the wsdd repo it may be better to wait for Bookworm otherwise we get stuck with any breakage that happens.

TheOldPresbyope, 2023-04-01:
I think this is a sensible position. Am in Vail CO at the moment to attend a wedding. When I get home I may create a (temporary) HowTo for those intrepid souls who can’t wait.

moodeaudio, 2023-04-15:
Decision to wait until Bookworm

closed 2023-04-15 by moodeaudio