moode-player/moode #612: No SSH daemon running on new installation

kannes, 2023-03-26:
I just installed the latest version available via `rpi-imager` on a Pi 3 B+, set up the network via AP mode (using the `moodeaudio` password without any initial setup) and made it connect to my wifi.

Now I want to configure some lowlevel things on the Pi but there does not seem to be a SSH daemon running. https://github.com/moode-player/moode/blob/master/www/setup.txt says "To access the operating system command console use Secure Shell (SSH)." so it sounds like it should be possible. I used it in the past all the time.

nmap only shows ports 80 and 111 being served. Trying to connect to port 22 gets refused.

The SECURITY section of https://github.com/moode-player/moode/blob/master/www/setup.txt says
> The ISO image does not contain the userid pi, SSH service, WiFi SSID or Access
> Point password.

The Web UI has a Web SSH option, which I enabled, its browser shell does not let me login using the old pi/moodeaudio (as the readme says) and there is no info on the user used now.

moodeaudio, 2023-03-26:
The security section in the guide specifies using the Raspberry Pi Imager to create the pi userid/password. The screen where you do this has a checkbox for "Enable SSH"

kannes, 2023-03-26:
Sorry, I must have accidently deleted a line from my post: There is no cog icon in rpi-imager when choosing moodeaudio:

[image]

The cog does appear for the official Pi OS:
[image]

moodeaudio, 2023-03-26:
What OS are you running the imager on?
Which imager option did you use to select the image file?

TheOldPresbyope, 2023-03-26:
Wait! This occurs for me too when I select "moOde Audio Player" from the "Other specific-purpose OS" list.

OTOH, if I scroll down to "Use Custom" and then select either the moOde image zip file or the unzipped moOde .img file, things works as I would expect. Once I've selected either of these choices the COG shows up. What is displayed in the Operating System box is the name of the file I chose, and not "MOODE AUDIO PLAYER (64-bit)".

kannes, 2023-03-26:
Ah nice, I will try it with a pre-downloaded image tomorrow. Thanks so much for the quick responses :)

moodeaudio, 2023-03-26:
I'm able to repro.

Reference this issue
https://github.com/raspberrypi/rpi-imager/issues/543

moodeaudio, 2023-03-26:
The Imager should now reflect the following updates:

- moOde appears in the Media Player OS category
- The COG icon shows up after selecting an image

picov, 2023-06-08:
I've just installed 2023-04-21-moode-r832-lite.img on my RPI4 and the device not answer on port 22.
I've enabled Web SSH but the default user/pass (pi/moodeaudio) not works.
 
Side note: I'm using the legacy 32 bit versione because the 64 bit version hang when I change network settings.

moodeaudio, 2023-06-08:
Troubleshooting is handled in the Forum. The Support page at moodeaudio.org has info on requesting a Forum account.

RichVarney, 2023-06-08:
> I've just installed 2023-04-21-moode-r832-lite.img on my RPI4 and the device not answer on port 22. I've enabled Web SSH but the default user/pass (pi/moodeaudio) not works.
> 
> Side note: I'm using the legacy 32 bit versione because the 64 bit version hang when I change network settings.

@picov I have just done the same thing. Then I re-read the guide as per the above https://github.com/moode-player/moode/issues/612#issuecomment-1484152400, used the Raspberry Pi imager and it worked with no issue.

I am just so used to flashing it with dd that I never attempted to use the official tools and only now did I read the docs.

moodeaudio, 2023-06-08:
You are not being forced to use the Pi Imager, it just makes prepping the image easy. The Setup Guide contains information on how to prep the image manually if you want to use some other image writer.

closed 2023-06-08 by moodeaudio

mathieuruellan, 2023-09-25:
> Troubleshooting is handled in the Forum. The Support page at moodeaudio.org has info on requesting a Forum account.

Ok but registration is closed :(