moode-player/moode #613: Local Display not working

Wandereradi, 2023-03-30:
WaveShare 7" DSI displayy fails to show the local display. 

Did a clean re-install and rebooting the system after setting the local display slider to "on" and clicking on Set,  no effect.
Did try to Retart local display- no display just stays on Verbose 

this display was functional on the MOODE prior to 8.3.0

Moode 8.3.0 64bit, RPI 4 - 1Gb

moodeaudio, 2023-03-30:
Your best bet is to contact the manufacturer of the display for help with its settings since we only test with the official Raspberry Pi Touch display.

You could also try posting in our Forum. Visit the Support page at http://moodeaudio.org for information on requesting an account.

closed 2023-03-30 by moodeaudio

moodeaudio, 2023-03-30:
Visit the Support page at http://moodeaudio.org/ for information on requesting an account.