moode-player/moode #614: Auto-trust bluetooth devices.

aero31aero, 2023-03-30:
Hi, I'm not sure if you're already aware of this, but apparently Linux's bluetooth stack itself doesn't play well with 'untrusted' devices. This has been my experience back with my old laptop as well as now with Moode.

Without trusting the devices, my phones aren't able to reliably connect to Moode's bluetooth despite having been successfully paired. I didn't dig deep into the logs so I don't know what the exact errors are, but on the phone end, there are no error messages; they just fail to connect.

I've been using a particular fix for ~2 years now which I've now had to redeploy on my new installation too, so I thought I'd share:

```

$ tail -n +1 auto-trust*
==> auto-trust.service <==
[Unit]
Description=Auto trusts all paired bluetooth devices.
Wants=auto-trust.timer

[Service]
Type=oneshot
ExecStart=/home/pi/auto-trust.sh

[Install]
WantedBy=multi-user.target

==> auto-trust.sh <==
#! /usr/bin/env bash
for mac in $(sudo bluetoothctl devices | awk '/^Device/ {print $2}'); do sudo bluetoothctl trust $mac;done

==> auto-trust.timer <==
[Unit]
Description=Auto trust bluetooth devices.
Requires=auto-trust.service

[Timer]
Unit=auto-trust.service
OnUnitInactiveSec=10s
OnBootSec=10s

[Install]
WantedBy=timers.target

```

aero31aero, 2023-03-30:
> IIRC the Pairing Agent option in Renderer Config does this already.

FWIW I had the pairing agent enabled, and restarted it and the device to
see if that'd solve it but it was still having issues.

That's when I remembered I'd applied this hack and reapplied it and it's
working fine again.

moodeaudio, 2023-06-22:
Moved to support Forum
https://moodeaudio.org/forum/showthread.php?tid=5648&pid=47303#pid47303

closed 2023-06-22 by moodeaudio