moode-player/moode #616: Three shell scripts contain spurious null characters

TheOldPresbyope, 2023-04-05:
A quality-control issue:

In moOde 8.3.0 and in the current repo, the following scripts (relative to the root of the repo) contain null characters
```
./home/piano.sh
./www/util/sysinfo.sh
./www/util/automount.sh
```

I uncovered this accidentally using the Linux `file` command in the moOde user's home directory on a working 8.3.0 player:
```
pi@m830p3a:~ $ file piano.sh
piano.sh: Bourne-Again shell script executable (binary data)
```

where the 'binary data' qualifier was a surprise; I would expect, instead, something like `Bourne-Again shell script, ASCII text executable`.

There's two reasons this could occur, but my sweep of these three shell scripts shows in each case the reason is the presence of one or more null characters at the ends of some lines, possibly the result of some misguided editing.

In Linux, the simplest way to expunge them is with the `tr` command
```
tr -d '\000' < original-file > repaired-file
```

moodeaudio, 2023-04-05:
Good catch. Fixed in latest commits.

It's happened before and I've never been able to track down the cause but I think it may be from copy/paste lines from the shell ??

moodeaudio, 2023-04-15:
Fixed.

closed 2023-04-15 by moodeaudio