moode-player/moode #627: Cover images not loaded

haecksenwerk, 2023-05-06:
I have created a music database consisting of music albums that live in separate directories on a USB stick, that is attached to the Raspberry that runs the Moode player. A cover art image is embedded in each mp3 file of the albums, but almost 50% of the albums do not show an image in the album view.

When taking a look into the `/var/local/www/imagesw/thmcache` folder, one can see that placeholder images are stored for the files from which the covers could not be loaded.

Since the covers show up in other players like VLC I. e., I wonder if there are some criteria that prevent certain covers from being pulled from the mp3 files?

moodeaudio, 2023-05-06:
I only use Git for developer related troubleshooting. General support is handled in our Forum so that all our users can view/participate in the discussion. Email tim at moodeaudio dot org with your preferred userid (at least 5 characters) and I'll setup an account for you.

closed 2023-05-06 by moodeaudio

haecksenwerk, 2023-05-06:
That makes sense. I'll write an email then. Thanks!