moode-player/moode #630: New release version is not seen from player

sstefanov, 2023-06-10:
Moode player is version 8.3.2. When check for new version there is a message:
Software is up to date
New release 8.3.3 is existing here, bit player cannot see it.
Here is part of system information page:

S Y S T E M   P A R A M E T E R S

moOde release		= 8.3.2 2023-04-21
RaspiOS			= 11.6
Linux kernel		= 6.1.21-v8+ #1642
Platform		= Pi-3B+ 1.3 1GB
Architecture		= aarch64 (64-bit)
Home directory		= /home/moode

sstefanov, 2023-06-10:
After little investigation seems that file with latest version is not available. File URL is:
https://raw.githubusercontent.com/moode-player/updates/main/moode-player

moodeaudio, 2023-06-10:
Official release announcements happen in the Forum

closed 2023-06-10 by moodeaudio

sstefanov, 2023-06-10:
In the Forum version 8.3.3 is announced as available.
The problem is that moode player cannot find it.

moodeaudio, 2023-06-10:
You prolly need to clear Browser cache or reboot Router.

sstefanov, 2023-06-11:
Now it's OK. Thank you!