moode-player/moode #632: worker don't start on rpi4

dmd79, 2023-06-16:
Hello, since moode 8.3.1 I have same issue about worker

pi@moode:~ $ cat /var/log/moode.log
20230616 233118 worker: --
20230616 233119 worker: -- Start moOde 8 series
20230616 233119 worker: --
20230616 233119 worker: Successfully daemonized
20230616 233119 worker: Waiting for Linux startup...
20230616 233247 command/index.php: Connection to MPD failed
20230616 233258 command/index.php: Connection to MPD failed
20230616 233439 worker: ERROR: Linux startup failed to complete after waiting 180 seconds

running on RPI4 2gb

moodeaudio, 2023-06-16:
Troubleshooting is provided in the Support section of our Forum so that all our users can benefit from the discussion. 

Let me know what userid you would like to use (at least 5 characters) and I’ll create an account for you.

closed 2023-06-16 by moodeaudio

dmd79, 2023-06-16:
dmd79
please give me the link of the forum
Thank

moodeaudio, 2023-06-17:
Sure, but I need a legit email address in order to setup an account.

Just send it to my email: tim at moodeaudio dot org and I'll reply with the account login info.

dmd79, 2023-06-17:
Sorry for misundertand but I've solved
the issue is not moodeaudio itself, but mpd_oled I needed last two commit
https://github.com/antiprism/mpd_oled/commits/master

moodeaudio, 2023-06-17:
Nice. You should still get a Forum account because I don't do any troubleshooting in Git, only developer issues.