moode-player/moode #639: Shutdown shortly after logging in

privatedev11, 2023-08-19:
Hi,
I have gotten moode on a RPI3B+, and I'm having an issue. Shortly after booting (aaprox 1 min), the system shuts down and is unresponsive. This has also happened on my RPI1 with the legacy version. It happens with ethernet and WLAN. I don't know why this would happen. Any help appreciated, thanks :)
Side note: 
I understand this should go in the forum, but I can't access this as I don't have an account and my request hasn't been read.

moodeaudio, 2023-08-19:
Visit the Support page at https://moodeaudio.org for info on easily requesting a Forum account. Also read the Setup Guide for info on configuring the image before writing it to boot media.

closed 2023-08-19 by moodeaudio

privatedev11, 2023-08-20:
Hi @moodeaudio 
This has worked for me, thank you! I need to read the guides next time XD
Also, I did request a forum account around a week ago but with no response, how long is the normal response time?
Thanks!

moodeaudio, 2023-08-20:
Usually I send the account request replies within the hour.

Go ahead and email tim at moodeaudio dot org and I'll create an account for you. Be sure to check your junk mail folder for the reply just in case.