moode-player/moode #644: Multi-language support --- contribute

onword, 2023-11-04:
I couldn't find the link to join the forums and post this, but at https://moodeaudio.org/forum/showthread.php?tid=5581 there is a conversation about multi-language support. It seems it is in development. I'd like to contribute Japanese strings (15+ years speaking and live with native speakers).

Please let me know if I can contribute and how to register for the moodeaudio forum to post there instead.

Thank you.

moodeaudio, 2023-11-04:
View the support page at http://moodeaudio.org for info on how to easily request a Forum account.

moodeaudio, 2023-11-14:
Local language development, support and work effort is explained in the relevant Forum post. I'll go ahead and close this issue but can reopen if needed.

closed 2023-11-14 by moodeaudio