moode-player/moode #649: cannot set network with an @ in it

fredericrous, 2024-01-11:
We cannot set a network with an @ in it from the ui. But it works if I set the network via Raspberry PI Manager

## steps to reproduce

- go to page `/net-config.php`
- under the section `Network (SSID)`, click the button "Enter" SSID manually
- a modal opens with the title Manual SSID Entry, enter a network name with an @ in it like "@home"
- click "Ok"

you get an error, only some characters are allowed, @ is not in it

## workaround

edit the input field and add an @ to the attribute "pattern" like so:

`pattern="[@A-Za-z0-9\-_ .]{3,32}"`

moodeaudio, 2024-01-11:
I'll add to the TODO list for upcoming release 8.3.8 (Q1 2024).

closed 2024-01-11 by moodeaudio

moodeaudio, 2024-01-11:
Fixed in commit https://github.com/moode-player/moode/commit/c0e4db7b45633b0edda9f9cf8f6278787fc60293