moode-player/moode #656: Expand API

CodaCodalis, 2024-03-10:
I plan to write an integration for moOde in Home Assistant. If the API would be expanded by Pause/Play that'd be great!

moodeaudio, 2024-03-10:
Which API are you referring to?

CodaCodalis, 2024-03-10:
The REST API mentioned in the [setup guide](https://github.com/moode-player/moode/blob/develop/www/setup.txt).

moodeaudio, 2024-03-10:
What command(s) are missing from the API for your usage scenario?

CodaCodalis, 2024-03-11:
My first draft for a moode-audio-card on home assistant (HA) dashboard contains a volume control (+/-) and displays the current volume. Essential feature would be to pause/resume the playback of the current title/radio station/playlist:

already available: get_volume, set_volume

missing: pause_playback, start_playback

If I get this running on HA I would love to display the title/artist on the card and some background image (similar to the Kodi integration for HA where fanart of the currently running track/movie is displayed on the card).

missing in that case: get_cover/get_fanart

moodeaudio, 2024-03-11:
MPD commands are supported for example play/pause/stop. Refer to Setup Guide
```
MPD commands
See MPD protocol for list of commands.
https://mpd.readthedocs.io/en/latest/protocol.html
```
Use command `get_currentsong` to return metadata and cover URL.

CodaCodalis, 2024-03-16:
Thanks. Will try that.

closed 2024-03-16 by CodaCodalis