moode-player/moode #660: Option to change the default state (pull-up / pull-down) of GPIO pins

ralphcrutzen, 2024-04-22:
Current situation: GPIO pins are always configured as input/pull-up.
Desired situation: It would be nice to see an option to change the default pin state to pull-down on the GPIO control page.
Motivation: With some buttons (like some rotary encoders with built-in buttons), the pin is connected to Vin when pushed, so it needs to be connected to a pull-down GPIO pin on the Raspberry Pi.

moodeaudio, 2024-04-22:
Should the option apply to all pins or should each pin have an option to configure as pull-up or pull-down?

ralphcrutzen, 2024-04-23:
Each pin is used for a different peripheral, so it would make more sense to set up each pin individually.

moodeaudio, 2024-05-04:
This has been implemented in upcoming moode 9

closed 2024-05-04 by moodeaudio