moode-player/moode #662: [FR] Add ChromeCast support

papampi, 2024-06-10:
So there is a [mpd2chromecast](https://github.com/dresdner353/mpd2chromecast ) which supports adding ChromeCast to Moode, why not  implement it into Moode UI?

papampi, 2024-06-10:
> Not enough project bandwidth but if you want to invest your time doing the development, integration and testing I can help out a bit.

Fair enough ... 
The install script seems to be outdated and won't installed on my rpi4 moode, will check it out and if I could fix will get to you for next steps.

moodeaudio, 2024-06-10:
I was referring to a dev doing 90% of the work to integrate this into moode as a renderer, not just helping you get it working on your own system.

papampi, 2024-06-10:
Sure, I can help with that too, if you needed tester or anything else I could do.

Just updated the install script to work on newer moode installs, you can check it here: [mpd2chromecast](https://github.com/papampi/mpd2chromecast)
mpd2chromecast Service is running ok now, set the http stream settings as well but get no cast on my cast receiver devices.

So if you needed me to do testings or anything else let me know.

papampi, 2024-06-11:
@moodeaudio 
So ... 
Successfully managed to run MPD2Chromecast and cast to my Nvidia Shield TV, added a simple section in Moode audio section too (thanks to [@5AMsan](https://github.com/5AMsan) for his comment in https://github.com/dresdner353/mpd2chromecast/issues/4).
Let me know how do you prefer to integrate it and I'll do whats needed.

[image]

moodeaudio, 2024-06-11:
I see it's more or less a stand-alone audio distribution component and not an audio renderer/receiver like AirPlay or Spotify Connect. It's basically side-loaded with its UI running in an iframe. Thats an order of magnitude less work than integrating a new audio renderer. Just have a look at the 2 recent Plexamp commits to get a sense of how many moving parts there are.

I think you should post a Thread in the forum describing the component and see what kind of responses come back. Email tim at moodeaudio dot org with your preferred userid and I'll create a Forum account for you.

papampi, 2024-06-11:
Acually I was looking for a Chromecast reciever for moode and thought the mpd2chromecast can do it, but after setting it up found out it's sending to Chromecast and it's impossible to have Chromecast reciever as it's closed source.
Don't think having moode stream to another Chromecast has so much usage.
Thanks for your awesome Moode.

closed 2024-06-11 by papampi

moodeaudio, 2024-06-11:
Wow what a waste of my time.