moode-player/moode #668: Support UPNP browsing in the UI

fr2lancer, 2024-07-06:
It would be good if Moode can provide the browsing function to remote / local upnp server and render them in the local  mpd or remote one.

As I like the UI and to use it as a consolidated browsing for all of sources.

moodeaudio, 2024-07-06:
From 3 years ago: release notes for 2022-06-21 moOde 8.1.0 (Bullseye)

`- UPD: Remove UPnP browser (djmount) which is unsupported on 64-bit RaspiOS`

The djmount repo has been unmaintained for almost 20 years

fr2lancer, 2024-07-06:
Djmount seems to be sshfs for upnp. Is fliesystem required to be as a input source to mpd or Moode?

moodeaudio, 2024-07-06:
Djmount was UPnP -> FUSE

The plugin you references has no origin source, license, authors or any other FOSS required parts.
I have no idea what the other async_upnp_client project does that you referenced

closed 2024-07-06 by moodeaudio