moode-player/moode #671: Timed CoverView display during playback

fabitom, 2024-07-19:
Hey! It's possible that I don't fully understand the idea of the coverview/screensaver, especially on a local LCD, but I think there should still be an option for the screensaver to work during playback, since we have a playbar that can be shown on the screensaver.
From the description, it seems it will only be active when MPD is not playing and the renderers are deactivated.
Currently, I'm testing unlocking the screensaver and I don't see any issues with the screensaver during playback.
I can make a PR if you want.

moodeaudio, 2024-07-19:
CoverView usage scenarios:
1. Timed display
This is classic screen saver mode. It's automatically activated after the timeout period when there is no playback activity (MPD or renderers). It's typically displayed w/o the Playbar.
2. Automatic display
This option supports the usage scenario where the Pi is connected to a large display at the front of the listening area, usually in-between the speakers/audio gear, and playback is controlled by Smartphone/Tablet at the listening chair. Timed display would be set to Never and Playbar set to No in this case.
3. Alternate main Playback view
This is supported by clicking on the TV icon. Timed display would be set to Never and Playbar set to Yes in this case. MPD can be playing. Timed display could also be set to a value but there has to be no MPD playback activity for it to take effect.

When a renderer connects its overlay is always displayed to prevent so user has access to the Disconnect/Turn off buttons. Access to the main WebUI is blocked to prevent changing any settings that would break the renderer config.

I think you are proposing a modification to Timed display whereby it would activate even if MPD is playing, correct?

fabitom, 2024-07-19:
Yes, exactly. I often listen to music before and during sleep ;) so having the screensaver with a clock left on overnight would be a nice option.

moodeaudio, 2024-09-17:
Will reopen if there is new information posted

closed 2024-09-17 by moodeaudio