moode-player/moode #675: Headphone / Jack not present in Audio output choices

TheFax, 2024-07-21:
I would like to use the Raspberry PI Jack audio output for listen music, but "Jack/Headphone" is not present on the list.
It would seem it should be present by reading the line above that says _"Select from Pi HDMI/Headphone, I2S or USB audio devices"_.
How to enable it?
Solution already tried:
- I tried to set a volume bigger than zero (like mentioned in another closed issue)
- I tried to reboot Moode disconnecting HDMI cable

[image]

[image]

moodeaudio, 2024-07-21:
Menu, Configure, System
Integrated audio
**Kernel mode:** Default vc4-kms-v3d driver for advanced Pi HDMI capabilities.
**Firmware mode**: Legacy snd-bcm2835 driver to support Pi Headphone only configuration. No other audio, video or system features are supported.

moodeaudio, 2024-07-21:
The issue tracker in this repo is for development related issues, not general support questions. 

If you have further support related questions, request an account on our Support Forum. The link is on the Support page at moodeaudio.org

closed 2024-07-21 by moodeaudio

tbrodbeck, 2024-08-19:
I have the same issue. For me this seems like a bug - why is HDMI recognized but the headphone jack not?

secnigma, 2024-12-25:
The same issue is happening intermittently with my setup, where the **Pi Headphones** Audio output goes missing randomly , even if **Firmware Mode (legacy)** is selected.

### **My Configuration:**
**_Hardware :_** Raspberry pi 3b+ 
**_Power Supply:_** Custom Power circuit built around [Hilink HLK-PM01 5v AC-DC SMPS ](https://jlcpcb.com/partdetail/hilink-HLKPM01/C209903)
**_OS:_** RPiOS 12.8 Bookworm 64-bit | Linux: 6.6.62 64-bit
**_MoodeAudio Version:_** 9.0.0
**_Audio output:_** Integrated 3.5mm Headphones Jack in Pi 3b+
**_SDCard_** : 16GB Sdcard (generic). 30% usage
**_Memory Usage_** : 174M/907M

### Current workaround 

To fix this issue, currently the workaround is to do the following:

1. _Go to Moodeaudio WebUI -> Menu -> Configure -> System -> Integrated Audio_

_Change  the mode to **Kernel Mode (default)** from **Firmware Mode (legacy)**_ 

[image]

2. _Reboot_

3. _Go to Moodeaudio WebUI -> Menu -> Configure -> System -> Integrated Audio_

_Change  the mode back to **Firmware Mode (legacy)** from **Kernel Mode (default)**_ 

4. _Reboot once again_

5. _Go to Moodeaudio WebUI -> Menu -> Configure -> Audio -> Audio Output -> Output device_

Now the missing **Pi Headphones** menu should be back.

[image]

moodeaudio, 2024-12-25:
Questions and troubleshooting for a release are handled in the Support section of our Forum so that all our users can benefit from the discussion.

Visit the Support page at moodeaudio.org for information on how to easily request a Forum account.