moode-player/moode #677: Add support for multichannel I²S input/output for RaspberryPi 5

renne, 2024-08-13:
On the RaspberryPi 5 it is possible to configure 4 GPIOs as I²S-inputs and 4 GPIOs as I²S outputs.
This allows to have 8 input and 8 output audio channels.

HiFiBerry already offers the [HiFiBerry DAC8x](https://www.hifiberry.com/shop/boards/hifiberry-dac8x/).

Combine the DAC8x with a HDMI audio-extractor with I²S output and you get a surround-system with active cross-over and frequency-response correction via CamillaDSP. 😁 

Please add the necessary overlays and UI configuration to MoOde! 😃

moodeaudio, 2024-08-13:
What are the overlays and what does the UI you want me to add look like?

renne, 2024-08-13:
The discussion about the overlays gets interesting at [this post](https://www.audiosciencereview.com/forum/index.php?threads/multichannel-audio-on-a-pi-will-get-a-whole-lot-easier-and-cheaper.48233/post-1861390).

For the UI it would be great to be able to select the HiFiBerry DAC8x and/or plain 8-channel I²S input/output in the list of audio devices. It should also be possible to select external MCLK on an additional GPIO on the RPi5 and switching between master- and slave-mode.

A [HDMI audio-extractor](https://de.aliexpress.com/item/33031920347.html?spm=a2g0o.order_list.order_list_main.10.24625c5fB9pzHw&gatewayAdapt=glo2deu) + MoOdle on RaspberryPi 5 in I²S slave mode + HiFiBerry DAC8x would be so great! :grin:

moodeaudio, 2024-08-13:
It looks like a very exotic solution requiring a lot of low level programming and config.
Its not something our project could support or troubleshoot.

moodeaudio, 2024-08-26:
I'll reopen if there is more information or interest.

MayorBug, 2024-09-16:
Just found out about the DAC8x and some research led me here. Call me interested

moodeaudio, 2024-09-17:
I think hifiberry is also working on an 8ch ADC
https://github.com/raspberrypi/linux/commit/4d2eaa194d77588fa42567ba174c3c14c5798027

You might want to ask them if this board can be used with their 8ch DAC board

closed 2024-11-01 by github-actions[bot]