moode-player/moode #678: Support album art and basic controls for spotify

alfredvaa, 2024-08-26:
Is there any plans on supporting album art, song title and maybe also basic controls for when streaming through Spotify?

moodeaudio, 2024-09-17:
Will reopen if there is new information posted

sm00th, 2024-09-20:
Hi, from what I saw, a recurring approach to this is to leave librespot to play audio and then use spotify's WebAPI to fetch metadata(e.g. https://github.com/Rigellute/spotify-tui) . This is not very user friendly: the user will have to create an "app" in spotify's developer dashboard (https://github.com/Rigellute/spotify-tui?tab=readme-ov-file#connecting-to-spotifys-api), auth and even then it will only show metadata for the authenticated user, not just anyone streaming to moode endpoint. But it is better than nothing and covers the most common usecase.

I've cobbled together a very rough PoC https://github.com/sm00th/moode/commit/9018f53f9e75766c5ae14a994ff084f6c8bbd018 that is able to show some metadata on the "Spotify Active" overlay. I think with some pointers on how to properly integrate this I can shape it into something usable.

Do you think this angle is worth pursuing?

moodeaudio, 2024-09-20:
Could be problematic since getting these dev API authentication tokens requires agreeing to legal terms.

sm00th, 2024-09-20:
Yes, but since we are using end-user tokens it is that user who accepts terms and conditions and decides whether they comply. So any possible liability also lies with that user and not the project.

moodeaudio, 2024-09-20:
Maybe, but I'm not a lawyer.

In any case the main issue with deep integration of a feature into moode code is that our project becomes responsible for ongoing support and maintenance and so I need to be reasonably sure that a fair number of users are going to use it. Most of our users are not devs or techs and so the process of getting Spotify API tokens might be a barrier to use.

Is there a way you can provide an installer or an image that users can test out? You could post a Thread about it in our Forum and see how the discussion goes.

sm00th, 2024-09-21:
Right, I didn't think about the fact that most userbase is non-technical. I am sure I'll be able to build an image, but before that I'd try to iron-out some sharp corners, right now it is to raw to show to non-devs I think.

moodeaudio, 2024-09-21:
You might want to do an installer/remover which would prolly be easier than a whole test image

Something like below (not tested). You would need to upload the two shell scripts and lib.min.js file to your repo. You could also have just one shell script with -install and -remove options.

__User download and run__
```
wget -q https://some_location_in_repo/install_spotmeta.sh -O ./install_spotmeta.sh
wget -q https://some_location_in_repo/remove_spotmeta.sh -O ./remove_spotmeta.sh
sudo chmod +x ./*spotmeta.sh
sudo ./install_spotmeta.sh
```
__install_spotmeta.sh__
```
echo "Installing 2024-MM-DD metadata update for Spotify Connect"
# Save original files
# If dir already exists skip this step so script can be run to install updated files and retain originals
if [ ! -d ./spotmeta-backup ]; then 
    mkdir ./spotmeta-backup
    cp  /var/www/ren-config.php ./spotmeta-backup
    cp  /var/www/inc/autocfg.php ./spotmeta-backup
    cp  /var/www/js/lib.min.js ./spotmeta-backup
    cp  /var/www/templates/ren-config.html ./spotmeta-backup
fi
# Install updated files 
moodeutl -q "INSERT OR IGNORE INTO cfg_system (id, param, value) VALUES (176, 'spotifyclientid', '')"
moodeutl -q "INSERT OR IGNORE INTO cfg_system (id, param, value) VALUES (177, 'spotifyclientsecret', '')"
moodeutl -q "INSERT OR IGNORE INTO cfg_system (id, param, value) VALUES (178, 'spotifyredirecturl', '')"
mkdir /var/www/spotify
wget -q https://github.com/sm00th/moode/blob/spotify/www/ren-config.php -O /var/www/ren-config.php
wget -q https://github.com/sm00th/moode/blob/spotify/www/inc/autocfg.php -O /var/www/inc/autocfg.php
wget -q https://github.com/sm00th/moode/blob/spotify/www/js/lib.min.js -O /var/www/js/lib.min.js
wget -q https://github.com/sm00th/moode/blob/spotify/www/spotify/index.html -O /var/www/spotify/index.html
wget -q https://github.com/sm00th/moode/blob/spotify/www/templates/ren-config.html -O /var/www/templates/ren-config.html
echo "Update successfully installed, reboot to make the changes effective."
```
__remove_spotmeta.sh__
```
echo "Removing 2024-MM-DD metadata update for Spotify Connect"
# SQL
moodeutl -q "DELETE FROM cfg_system WHERE param='spotifyclientid'"
moodeutl -q "DELETE FROM cfg_system WHERE param='spotifyclientsecret'"
moodeutl -q "DELETE FROM cfg_system WHERE param='spotifyredirecturl'"
# Session vars
moodeutl -D spotifyclientid
moodeutl -D spotifyclientsecret
moodeutl -D spotifyredirecturl
# Files
cp  ./spotmeta-backup/ren-config.php /var/www/
cp  ./spotmeta-backup/autocfg.php /var/www/inc/
cp  ./spotmeta-backup/lib.min.js /var/www/js/
cp  ./spotmeta-backup/ren-config.html /var/www/templates/
# Dirs
rm -rf /var/www/spotify
rm -rf ./spotmeta-backup
echo "Update successfully removed, reboot to make the changes effective."
```

sm00th, 2024-09-25:
It seems lib.min.js is generated so is unavailable in the repo. There also might be other problems if these files, which are based on 9.1.2 are copied over a stable (9.1.0) install. Package creation is pretty straightforward and should be more reliable, so I'll go with that and database update instructions along it.

closed 2024-11-10 by github-actions[bot]

pbattino, 2024-11-13:
I'm probably missing something but... Isn't it possible to get notified directly from librespot using the --onevent=myscript.sh parameter and then reading metadata from environment variables?

See doc
https://github.com/librespot-org/librespot/wiki/Events

https://github.com/librespot-org/librespot/blob/master/contrib/event_handler_example.py

moodeaudio, 2024-11-13:
This has already been implemented in upcoming moOde 9.1.5. If u want more info, screen shots, test release etc your best bet is to get a Forum account. 

Visit the Support page at moodeaudio.org for information on how to easily request an account.

pbattino, 2024-11-13:
Thank you! I'm on 9.1.4 , I did not check the upcoming release yet.
Great work!

moodeaudio, 2024-11-13:
Did u want a Forum account?

All the Q&A, support and troubleshooting is done in the Forum. I only use Git issues for Development related stuff.

pbattino, 2024-11-14:
> Did u want a Forum account?
> 
> All the Q&A, support and troubleshooting is done in the Forum. I only use Git issues for Development related stuff.

Yes please!!!