moode-player/moode #679: [Doc] Setup Guide link on Releases page doesn't exist (anymore ?)

typhoe, 2024-08-26:
Hi,

when you try to follow the link for the `Setup Guide` on the [Releases](https://github.com/moode-player/moode/releases), it points to [https://github.com/moode-player/moode/blob/master/www/setup.txt](https://github.com/moode-player/moode/blob/master/www/setup.txt)

There's no setup.txt file in this folder.
I guess the right link is now: [https://github.com/moode-player/docs/blob/main/setup_guide.md#setup-guide-](https://github.com/moode-player/docs/blob/main/setup_guide.md#setup-guide-)

Anyway, just to let you correct the link.

Thanks

moodeaudio, 2024-08-26:
Thanks. I'll fix but it will take some time.

When I'm finished with all the releases I'll post back and close the issue.

moodeaudio, 2024-08-26:
Looks like it's only necessary back to 8.3.7. Prior to that release the ISO images have been removed because of security issues so there's no point in changing the setup guide link in those releases. At some point they will prolly be removed from the Releases list.

All the releases from 8.3.7 -> 9.0.8 have been updated with the correct Setup Guide link.

I'll reopen if there are any more issues.

closed 2024-08-26 by moodeaudio