moode-player/moode #681: Spotify playback broken

prehensile, 2024-09-07:
Tracks just skip on through without playing. This has suddenly started happening at some point in the last couple of weeks, on an untouched installation of moode.

I strongly suspect it's caused by [this reported issue with librespot](https://github.com/librespot-org/librespot/issues/1322), given the coincident symptoms and timing.

The current workaround reported on the librespot issue is to switch to their current dev version. Any chance we can get a patch / hotfix / instructions to manually replace the version of librespot used by moode?

moodeaudio, 2024-09-07:
Support is handled in our Forum so that all our users can benefit from the discussion. Visit the Support page at http://moodeaudio.org for a link to request a Forum account.

closed 2024-09-07 by moodeaudio

prehensile, 2024-09-08:
> Support is handled in our Forum so that all our users can benefit from the discussion. Visit the Support page at http://moodeaudio.org for a link to request a Forum account.

I sent an email to request an account, but didn't get anything back. So tried here instead!

moodeaudio, 2024-09-08:
Maybe it ended up in a spam filter somewhere.

We need a legit reply email address in order to create an account.